RobotLocomotion/drake #7946: Polynomial library throws exception when integrating a constant polynomial

jefesaurus, 2018-02-03:
It seems like it doesn't have a default variable name and complains about it. e.g.:

```
TEST(PolynomialTest, Smoke) {
  // Ramp piecewise poly with one segment.
  std::vector<MatrixXd> controls;
  controls.push_back(MatrixXd(1, 1));
  controls.push_back(MatrixXd(1, 1));
  controls[0](0) = 0.0;
  controls[0](1) = 1.0;
  std::vector<double> breaks = {0, 1};
  auto piece_poly = PiecewisePolynomial<double>::FirstOrderHold(
        breaks, controls);
  // Ramp poly
  auto poly = piece_poly.getPolynomialMatrix(0)(0);

  // Derivative is constant.
  auto derivative = poly.Derivative();

  // Try to get back to the original ramp, chokes with: 
  // C++ exception with description "don't know the variable name" thrown in the test body.
  auto integrated = derivative.Integral(poly.EvaluateUnivariate(0));
}
```

hongkai-dai, 2020-12-09:
I will close this issue. I think the problem is that we need to pass another variable when doing `derivative.Integral` to specify which indeterminate is integrated. This problem shows up not only for a constant polynomial, but for polynomial with multiple indeterminates, like `x^2 * y^3`. When we integrate over `x`, the result is different from integrating over `y`.

hongkai-dai, 2020-12-09:
I read it wrong. This is not for a symbolic polynomial, but for a `Polynomial` class with single variable `t`.