RobotLocomotion/drake #7958: various tests spoil bazel cache by writing to current directory.

rpoyner-tri, 2018-02-05:
Try the following experiment on latest master:
```
 $ bazel clean
 $ bazel test //tools/workspace/lcm:install_test
 # test passes.
 # make some whitespace or comment change to tools/workspace/lcm/test/install_test.py
 $  bazel test //tools/workspace/lcm:install_test
 # test fails, with an error from external/bazel_tools/tools/test/test-setup.sh.
```
Various of the *install_test tests are writing to subdirs of the current working directory of the test sandbox, which is not safe in bazel, and is in fact formally forbidden by contract. Quoting from https://docs.bazel.build/versions/master/test-encyclopedia.html#test-interaction-with-the-filesystem : 

> No directory, file, or symlink within the runfiles tree (including paths which traverse symlinks) should
> be writable. (It follows that the initial working directory should not be writable.) Tests must not assume
> that any part of the runfiles is writable, or owned by the current user (i.e. chmod and chgrp may fail).

> The runfiles tree (including paths which traverse symlinks) must not change during test execution. 
> Parent directories and filesystem mounts must not change in any way which affects the result of 
> resolving a path within the runfiles tree.

Affected tests that I know of are:
* //bindings/pydrake:common_install_test
* //lcmtypes:drake-lcm-spy_install_test
* //tools/workspace/lcm:install_test

jwnimmer-tri, 2018-02-05:
Known issue, will be fixed by #7774.  If its an acute problem we could try to merge a partial fix in the meantime.

rpoyner-tri, 2018-02-05:
Well then. I'll just leave my patch here: https://github.com/rpoyner-tri/drake/commit/b28c9d328f5d672b6d8272b9fcd42a22a39d54cb . Let me know if you want me to help more on this.