RobotLocomotion/drake #7981: Python modules have explicit framework links on macOS

jamiesnape, 2018-02-07:
Relates #1183. This is the output of the `brew audit` command for a potential `libdrake` formula.

```
    These python extension modules were linked directly to a Python
    framework binary. They should be linked with -undefined dynamic_lookup
    instead of -lpython or -framework Python.
      /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/multibody/rigid_body_tree.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/multibody/rigid_body_plant.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/multibody/shapes.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/multibody/parsers.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/_common_py.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/solvers/gurobi.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/solvers/mosek.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/solvers/ipopt.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/solvers/_ik_py.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/solvers/_mathematicalprogram_py.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/examples/pendulum.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/_autodiffutils_py.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/_symbolic_py.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/systems/sensors.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/systems/analysis.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/systems/primitives.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/pydrake/systems/framework.so
            /usr/local/opt/libdrake/lib/python2.7/site-packages/lcm/_lcm.so
```

This is very common defect on macOS and the solution is exactly in the output, using `-undefined dynamic_lookup`.

FYI @EricCousineau-TRI.

EricCousineau-TRI, 2018-02-07:
Just to verify, does this still handle different Python versions / implementations robustly (e.g. PyPy, Python3 vs. Python2, etc.)?

I would assume so, since the generated modules tend to have more decorating information to indicate what implementation they're generated for (aside from Python2.7, that is).

jamiesnape, 2018-02-07:
Yes, and it handles system vs. Homebrew Python rather better.

PR #7983 (2018-02-08): Use -undefined dynamic_lookup when linking to python on macOS
Fixes #7981.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7983)
closed 2018-02-08 by jamiesnape