RobotLocomotion/drake #7982: Installed locations of vendored dependencies conflict with Homebrew installed versions

jamiesnape, 2018-02-07:
Relates #1183. Namely `eigen`, `bullet`, `fmt`, and `spdlog`. Following Homebrew conventions, we could install to `libexec/vendor/{include|lib}` or we could even just use system versions.

EricCousineau-TRI, 2018-12-03:
Merged into #1183.

closed 2018-12-03 by EricCousineau-TRI

jwnimmer-tri, 2020-02-28:
Per #11758 / #12780, we're going to use homebrew Eigen.  If homebrew offers fmt and spdlog then I'd be happy for use to use the homebrew version.

jamiesnape, 2020-02-28:
```
% brew info fmt
fmt: stable 6.1.2 (bottled)
Open-source formatting library for C++
https://fmtlib.github.io/
Not installed
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/fmt.rb
==> Dependencies
Build: cmake ✔

% brew info spdlog
spdlog: stable 1.5.0 (bottled), HEAD
Super fast C++ logging library
https://github.com/gabime/spdlog
Not installed
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/spdlog.rb
==> Dependencies
Build: cmake ✔
```