RobotLocomotion/drake #7984: Investigate SNOPT 7.5 new interface

hongkai-dai, 2018-02-07:
We used an old Fortran style `snopta_` API in SNOPT7.2. In SNOPT 7.5 they provide a new interface https://github.com/snopt/snopt-interface/blob/master/include/snopt.h. We should investigate using this interface, and compare the performance with the old one.

jwnimmer-tri, 2018-03-22:
The old f2c interface no longer supported by upstream.  We should definitely move to the new interface, when we have some time to port our code.

jwnimmer-tri, 2018-03-22:
Possibly #1667 could be resolved during this porting, also.

jwnimmer-tri, 2019-01-03:
There is now a method `SnoptSolver::is_thread_safe` to report whether the build of SNOPT wrapped into Drake is thread-safe or not.  To enable the thread-safe build, use `--config snopt_fortran` instead of `--config snopt` when compiling via bazel (not yet exposed through a CMake knob).

I think what remains for this issue is to either offer the CMake option for Fortran (or perhaps the Fortran version becomes the everywhere default), and to make sure we have CI coverage for it.

jwnimmer-tri, 2019-01-03:
As of #9896, we have support for this new interface, though it's disabled by default.  To enable it, use --config snopt_fortran instead of --config snopt when compiling via bazel (not yet exposed through a CMake knob).

I think what remains for this issue is to either offer the CMake option for Fortran (or perhaps the Fortran version becomes the everywhere default), and to make sure we have CI coverage for it.

jwnimmer-tri, 2019-01-16:
FYI I'm consolidating the work blocking such a transition in #10422.

jwnimmer-tri, 2019-01-16:
... and I think #10422 can subsume this issue entirely.  I'll close this one as a duplicate.

closed 2019-01-16 by jwnimmer-tri