RobotLocomotion/drake #7991: install_prereqs.sh Error

airfield20, 2018-02-07:
Before running this script i ran sudo apt update; sudo apt upgrade; sudo apt dist-upgrade
Then when following the build process I experienced this error.

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 coinor-libipopt1v5 : Depends: libmumps-seq-4.10.0 but it is not going to be installed
 dreal : Depends: coinor-libclp-dev but it is not going to be installed
 libboost-all-dev : Depends: libboost-date-time-dev but it is not going to be installed
                    Depends: libboost-filesystem-dev but it is not going to be installed
                    Depends: libboost-iostreams-dev but it is not going to be installed
                    Depends: libboost-log-dev but it is not going to be installed
                    Depends: libboost-python-dev but it is not going to be installed
                    Depends: libboost-system-dev but it is not going to be installed
                    Depends: libboost-thread-dev but it is not going to be installed
                    Depends: libboost-wave-dev but it is not going to be installed
 libhdf5-10 : Depends: libgfortran3 (>= 4.3) but it is not going to be installed
 libibex-dev : Depends: coinor-libclp-dev but it is not going to be installed
               Depends: zlib1g-dev but it is not going to be installed
 liblapack3 : Depends: libgfortran3 (>= 4.6) but it is not going to be installed
 libprotobuf-dev : Depends: zlib1g-dev but it is not going to be installed
 python-scipy : Depends: libgfortran3 (>= 4.6) but it is not going to be installed
 qtbase5-dev : Depends: libgl1-mesa-dev but it is not going to be installed or
                        libgl-dev
               Depends: libgles2-mesa-dev but it is not going to be installed or
                        libgles2-dev but it is not installable
               Depends: libglu1-mesa-dev but it is not going to be installed or
                        libglu-dev
               Depends: libqt5core5a (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5dbus5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5gui5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5network5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5printsupport5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5sql5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5test5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5widgets5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
               Depends: libqt5xml5 (= 5.5.1+dfsg-16ubuntu7) but 5.5.1+dfsg-16ubuntu7.5 is to be installed
E: Unable to correct problems, you have held broken packages.
The Drake source distribution prerequisite setup script has experienced an error on line 15 while running the command source "${BASH_SOURCE%/*}/install_prereqs_binary_distribution.sh"
```

airfield20, 2018-02-07:
When installing on a fresh install (without running apt upgrade or dist-upgrade) this issue doesnt occur

EricCousineau-TRI, 2018-02-08:
This might be something specific to your system setup, possibly due to other packages that have dependencies that conflict with Drake's. If you know what these conflicts are, can you post a link them in a GitHub Gist (to avoid cluttering the discussion)?

If you have things that you do not wish to break, you might want to consider using Drake in an isolated environment, such as Docker or a virtual machine.

The Drake instructions for Docker are in the process of being updated, but if you wish to just tinker with Drake, consider using a release version from Docker, which is done by the Underactuated class:
http://underactuated.csail.mit.edu/Spring2018/install_drake_docker.html

If you are OK with potentially disabling / breaking other packages / applications, you can try some of [these options](https://askubuntu.com/questions/563178/the-following-packages-have-unmet-dependencies) - but please be wary!

airfield20, 2018-02-08:
It was a fresh install of Ubuntu 16.04 apart from running apt get upgrade; apt get dist-upgrade. There were no other packages installed on the system that were not default packages.

jwnimmer-tri, 2018-02-08:
I have seen a similar thing if the Ubuntu sources lists were not wide enough.  In general if Xenial, Xenial Updates, and Xenial Security are all selected for all of main, universe, restricted, multiverse then things work.  If you subset that (especially after installing updates but then disabling them) the dependencies might not be available.

airfield20, 2018-02-08:
Ok, that may have been the issue, but I have no way of testing because I've already done another fresh install.

EricCousineau-TRI, 2018-02-08:
Given that this is not presently reproducible, I will close this for now. We can reopen it (and #8007) if it the problem crops up again.

Thanks for reporting it!

closed 2018-02-08 by EricCousineau-TRI

BryceeeeeChen, 2018-03-20:
Hi, I also encountered this problem when I tried to install Drake following readme in https://github.com/osrf/kumonoito (#8374). Enabling more sources lists did not solve the problem. I needed to clear all errors that would show up after ran `sudo apt update/upgrade`, even though those errors were irrelevant to Drake itself. Actually, the issue can be reproduced by simply adding an invalid software repository to the source list.