RobotLocomotion/drake #7994: RgbdCamera, VTK backend, introduces odd rendering artifacts

EricCousineau-TRI, 2018-02-07:
I've noticed these artifacts in `drake_visualizer` and did not think too much about it, but now it seems these same artifacts are occurring in `RgbdCamera`.
This also appears to affect the depth image, so may make it hard to use this for simulation.

[image]

@jamiesnape Any clues on why this might be happening?

(This is from a `lcmlog`, played by `@lcm//:lcm-logplayer-gui`, with a simulation running from a project downstream of Drake.)

@kunimatsu-tri I've assigned it to you for now, but please let me know if it should go to anyone else.

kunimatsu-tri, 2018-02-07:
Thanks @EricCousineau-TRI for filing the issue.

Can you elaborate more about what's happening here? Is the artifact that you got no depth return at all?

Also, can you let me know how to reproduce this including models you used?

SeanCurtis-TRI, 2018-02-08:
Just tagging myself so I can follow further conversation on this issue.

jamiesnape, 2018-02-08:
I am working on getting VTK 8.1 into Drake. Once that is in, I will revisit this. Can you elaborate or annotate the images so I know exactly to which artifact you are referring.

kunimatsu-tri, 2018-02-08:
I confirmed that the artifact happens on my local machine as well.

Here is the video: https://youtu.be/pHDSDDx89qs

In this video, you see the surface of table (grey) is flicking and occasionally turning into weird looking. And I saw the same symptom happens in `drake_visualizer` as well when I move the screen view angle by dragging the screen.

Unfortunately, I have no idea why this happens for now.

SeanCurtis-TRI, 2018-02-08:
This looks exactly like "z-fighting".  I'm going to make the following assumptions, tell me which are wrong:

1) You have stacked geometry -- i.e., two surfaces facing the direction in near proximity to each other.
2) Your range spanned by the near plane and the far plane is much larger than the distance from camera to surface.
3) You have limited depth precision (this one is the least likely factor - and countering this won't disprove z-fighting.)

EricCousineau-TRI, 2018-02-08:
Oh goodness, I believe it may have been due to us placing the table (the surface that's possibly z-fighting) below the ground, so (1) is the culprit. (I had plugged in the camera without looking too deeply at the physical layout.)

I will try moving the scene up.

EricCousineau-TRI, 2018-02-08:
Well, that's embarrassing. Yes, the z-fighting was just due to the ground plane and the table surfaces. I moved the scene up, and that fixed it. Hadn't noticed the issue in `drake-visualizer` because the ground had not been added (and I wasn't paying attention to the grid).

Sorry about that, and thanks!

closed 2018-02-08 by EricCousineau-TRI

SeanCurtis-TRI, 2018-02-08:
No worries. It's nice to have an easy one every once and a while.