RobotLocomotion/drake #8003: Promote `geometry/test_utils/expect_error_message.h` to `drake/common`

amcastro-tri, 2018-02-08:
There are some really useful testing utilities there that are worth promoting so everyone can benefit from them.
Now in use by:
 - `drake/geometry/` code.
 - `drake/multibody/multibody_tree` code, #7995.

amcastro-tri, 2018-02-08:
I just found out that `diagram_builder_test.cc` has the same macro. Rule of three satisfied now! List updated.

amcastro-tri, 2018-02-08:
BTW. @soonho-tri proposes ([see comment](https://reviewable.io/reviews/robotlocomotion/drake/7995#-L4qfFOJYbzSN0vlFPVq)) to append these macros with `DRAKE_`

soonho-tri, 2018-02-08:
FYI, this should be a complete list of the files using the macro:
```
$ ag "EXPECT_ERROR_MESSAGE" -l
common/test/copyable_unique_ptr_test.cc
common/test/type_safe_index_test.cc
geometry/test_utilities/expect_error_message.h
geometry/test/frame_id_vector_test.cc
geometry/test/query_object_test.cc
geometry/test/geometry_state_test.cc
geometry/test/geometry_system_test.cc
systems/framework/test/diagram_builder_test.cc
```

SeanCurtis-TRI, 2018-02-08:
Those files are like my breadcrumbs -- that paint a clear picture of which code I've been walking through.

Thanks, @soonho-tri for the thorough accounting. And I also support the `DRAKE_` prefix.

PR #8396 (2018-03-19): Promote DRAKE_EXPECT_THROWS_MESSAGE et al. from geometry to common
This PR replaces an assortment of bespoke macros for verifying in unit tests that an error message contains the expected text. Call sites are updated to use the now-standardized signatures. 

Semantics of the "IS_ARMED" forms has changed: as before the message is only _required_ in Debug
(actually `DRAKE_ENABLE_ASSERTS`), but is now _optional_ in Release (previously it was _forbidden_ in Release). _If_ the Release build actually throws, the error message must meet the same requirements as in Debug.

Renamed from EXPECT_ERROR_ME [... cut]
closed 2018-03-19 by sherm1