RobotLocomotion/drake #8010: drake_visualizer: Do not depend on `libdrake.so` by default, use stubbed `pydrake` for minimal functionality.

EricCousineau-TRI, 2018-02-08:
From Slack convo:

> jwnimmer-tri [2:34 PM]
> Am I correct on my reading that the dependency from `//tools:drake_visualizer` onto `"//bindings/pydrake"` is only to have `pydrake.getDrakePath()` and `pydrake.common.FindResourceOrThrow`?  It seems to me that we could weaken that dependency to avoid compiling a lot of C++ code that we don't need.
> 
> jamiesnape [3:11 PM]
> Primarily, but not sure if it is the only reason. There are other parts of `pydrake` used in the `director` repo.
> (I would say probably, though.)
> 
> eric.cousineau [3:34 PM]
> I'd be fine if we made two targets, one without `pydrake` and one with `pydrake`. Will whip up a PR once the other `drake_visualizer` PR lands.
> 
> jamiesnape [3:40 PM]
> We definitely need part of `pydrake` for the aforementioned functions. The question is how much else (if anything). (edited)
> 
> jwnimmer-tri [3:44 PM]
> Yeah.  My proposal would be to just change the `deps=[]` line of `//tools:drake_visualizer` to be a narrower thing, possibly even completely separate stubs for those two functions (when run from the sandbox).
> 
> jamiesnape [3:45 PM]
> Sounds good to me.

jwnimmer-tri, 2018-02-08:
I suspect this will be a pain point in Anzu sooner rather than later, so +(status: priority: high).  You can kick it over to me if you're short on time.

jwnimmer-tri, 2018-02-08:
(In the alternative, I guess Anzu could have its own launcher wrapper with different deps, but the benefit to Drake developers seems worth doing it directly in-tree here.)

EricCousineau-TRI, 2018-02-09:
I'm not sure if I'd get to this quickly, so I will take you up on your offer - thanks! (and boot it back if you get bogged down)

PR #8080 (2018-02-15): Remove drake_visualizer's dependency on pydrake
Fixes #8010.

I've tested this locally using:
- the `examples/vakyrie/README.md` instructions;
- the first example from `automotive/README.md` instructions;
- the Anzu demo instructions.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8080)
closed 2018-02-15 by jwnimmer-tri