RobotLocomotion/drake #8012: `drake.lcmtypes` does not have an `__init__.py`, thus cannot always be imported

EricCousineau-TRI, 2018-02-08:
Not sure when exactly this passes or fails, but `lcmtypes_py` does not have an `__init__.py`, so importing things breaks.

Trying to use `show_contacts.py` with `//tools:drake_visualizer`:
```
...
    messageClass=lcmdrakemsg.lcmt_contact_results_for_viz,
AttributeError: 'module' object has no attribute 'lcmt_contact_results_for_viz'
```

A solution is to place an `__init__.py` in `lcmtypes_py`.

\cc @kunimatsu-tri @amcastro-tri

jwnimmer-tri, 2018-02-08:
The codegen by default makes one.  Maybe its a drake-the-workspace vs drake-the-module confusion?

EricCousineau-TRI, 2018-02-09:
I don't think so - it's reading your `//:module_py` code to check the `_needs_lcmtypes_pathing` logic. When I inspected `.runfiles`, `.../drake_visualizer.runfiles/drake/lcmtypes` had no `__init__.py`.

If the codegen does generate an `__init__.py`, I think that's going in under `drake/lcmtypes/drake`.

jwnimmer-tri, 2018-02-09:
Yeah, I'd misunderstood which `__init__.py` you'd meant was missing.  I see what the PR is trying to do now; I'll give it some mental energy later on to see if it seems robust.

PR #8013 (2018-02-10): lcmtypes: Ensure module has `__init__`.
Closes #8012.

Since we want to avoid `add_current_package_to_imports` (which would have possibly generated an `__init__`, but cluttered PYTHONPATH), this provides a way to incorporate an `__init__.py`.

~An alternative would be to update `drake_lcm_py_library` to take an `extra_srcs` parameter. May do that right quick.~

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8013)
closed 2018-02-10 by jwnimmer-tri