RobotLocomotion/drake #8025: drake_visualizer: Ensure mesh caching picks up model updates

EricCousineau-TRI, 2018-02-09:
At present, `drake_visualizer` caches meshes that it loads.

A solution is to either (a) do not cache meshes (risking slowdown) or (b) store an mtime timestep on read for the filepath, and do a quick timestamp check.

@patmarion If you're available, any hints on quick fixes / flags for this?

\cc @sammy-tri @SeanCurtis-TRI @jwnimmer-tri @jamiesnape

jwnimmer-tri, 2018-02-09:
In `#visualization` it also came up that just sending the whole mesh over the wire may work well also.

SeanCurtis-TRI, 2018-02-09:
Sending the whole mesh requires a protocol change.

EricCousineau-TRI, 2018-02-09:
From `#visualization`:

```
sean.curtis [10:10 AM]
@jamiesnape follow up question. I get the sense that when I send an lcm message to load an obj file, that drake visualizer creates some cached version of it. I can no modify the contents of the original obj file, but it has no effect whatsoever on what is being loaded -- it persists in loading the data that it found the first time it loaded it. Is my guess correct? And where would I find that cached/modified beast?

jamiesnape [10:14 AM]
I think there is an internal conversion somewhere to a VTK polydata. I will check. To answer yesterday’s question, I forget to look in the Python, so here is the logic for MTL files: https://github.com/RobotLocomotion/director/blob/4109097ab03fe2728bf6ac9a9be1be952e449153/src/python/director/ioUtils.py#L91

sean.curtis [10:18 AM]
I just realized it's an in-memory representation.
By restarting drake_visualizer, I get a clean load of the current state of the file. So, all of my flailing had been for naught; drake_visualizer persisted in displaying the initial load regardless of all subsequent changes.

jwnimmer-tri [10:19 AM]
@sean.curtis "send an lcm message" -- which protocol?  lcmt_viewer_load_robot or https://github.com/RobotLocomotion/director/blob/master/docs/sphinx/user_guide/tree_viewer_interface.rst?

sean.curtis [10:19 AM]
`lcmt_viewer_load_robot` - i.e., the standard mechanism that causes drake to push its visual state over to drake_visualizer.

jamiesnape [10:20 AM]
@sean.curtis Yes, in memory as a VTK polydata.

sean.curtis [10:20 AM]
Sadly, it doesn't have a timestamp that it compares with the file -- leading to these woes.
But now I know.

rdeits [11:09 AM]
I've been doing a lot of work recently on the Remote Tree Viewer interface, and I think the way forward is just to stop sending meshes by filename entirely. Instead, I do all of my mesh visualization by sending the entire mesh over the wire. This was previously pretty inefficient, but I've just gotten a new ZeroMQ+MsgPack interface working and it's now fast enough that sending a mesh with thousands of vertices takes < 2ms (and pointclouds of up to 10,000,000 points work quite well too). Doing this solves any mesh staleness issues and also completely removes the need to have Director know anything about Drake's resources or `package://` paths.
I'm still considering using ProtoBuf instead of MsgPack, but MsgPack is just *so nice* for simple protocols like this (and `msgpack-numpy` is shockingly efficient at packing large arrays).

sean.curtis [11:18 AM]
Good to know -- for now, it's immensely convenient for me to pass the file name (it's a stopgap measure). But moving forward that has some nice appeal, particularly when we make Drake-side geometry modifications.

rdeits [11:22 AM]
Yeah, understood. The Tree Viewer still supports a `mesh_file` type (which just passes a filename), although it will suffer from the same staleness issues you've already noticed.
```

patmarion, 2018-02-10:
Here is the line of code where the mesh data is stored in the cache (python dict):

https://github.com/RobotLocomotion/director/blob/master/src/python/director/drakevisualizer.py#L229

Commenting out that line will disable the cache.

Here is the line where the cache is read:

https://github.com/RobotLocomotion/director/blob/master/src/python/director/drakevisualizer.py#L199

If you want to change the dict key from `filename` to `(filename, os.path.getmtime(filename))` that would be fine with me.  Actually, the diff would be slightly more complicated, because the filename is transformed in the lines just below, so you don't want the mtime of the original filename, but the transformed filename.

One comment about sending the mesh data over the wire: currently the mesh data is received in the python layer, not the c++ layer, so vtk mesh construction occurs in python.  This could be slow for large meshes.  For mesh filenames, c++ data readers are used.

jwnimmer-tri, 2021-12-14:
@EricCousineau-TRI I propose closing this as a"WONTFIX" per #16215.  What do you think?

SeanCurtis-TRI, 2021-12-14:
As with #14532, perhaps the "solution" is to document it clearly. A compromise between "won't fix" and work which yields ever decreasing value.

EricCousineau-TRI, 2022-01-12:
Will close for now, letting issue serve as documentation. Can hoist if a user hits snag again.

closed 2022-01-12 by EricCousineau-TRI