RobotLocomotion/drake #8045: IDM cars won't pass each other

airfield20, 2018-02-12:
I am having an issue where the error:  `abort: Failure at bazel-out/k8-opt/bin/systems/analysis/_virtual_includes/simulator/drake/systems/analysis/simulator.h:572 in StepTo(): condition 'next_sample_time >= step_start_time' failed.` occurs when an IDM car in one lane is about to pass an IDM car in an adjacent lane on a curved segment.

Also on the straight segments, if I have an IDM car in lane 1 and the IDM car in lane 2 is is traveling faster than the car in the first lane then it will slow down. I wouldn't expect the cars to be affected by other cars in adjacent lanes. 

Is #8028 related to my issue? If so, will it fix my issue?

An example main is attached.

[main2.txt](https://github.com/RobotLocomotion/drake/files/1717639/main2.txt)

jadecastro, 2018-02-13:
I checked your example over and I see the problem is that the `multilane` implementation is returning a `RoadPosition` that does not minimize `r`-position, as the API docs state. In particular, the IDM cars are placed either in `l:1_0` or `l:1_1`, but in _both_ IDM cars, `ToRoadPosition()` always returns `l:1_0` (with an `r`-offset of either 0 or 5). This defeats the PoseSelector computations.

The `ToRoadPosition` API docs state 
> If the chosen Segment has multiple Lanes, then ToRoadPosition() will choose a Lane which contains geo_position within its `lane_bounds()` if possible, and if that is still ambiguous, it will further select a Lane which minimizes the absolute value of the lateral `r` coordinate in the result

but according to the `multilane` implementation, the first qualifying lane found in the search is returned: https://github.com/RobotLocomotion/drake/blob/master/automotive/maliput/multilane/road_geometry.cc#L118

Keep in mind that the `multilane` implementation has not yet stabilized.

/cc @maddog-tri - if you're not already aware of this.

jadecastro, 2018-02-13:
/cc @agalbachicar

agalbachicar, 2018-02-28:
@airfield20 JFYI, we are currently validating a similar environment as the one you proposed above with a fix. As soon as we can finish testing it, I will open a PR and tag you so you can follow up the fix.

agalbachicar, 2018-03-05:
@airfield20 , the PR that fixes this issue is merged. Please, try it and let us know if there is anything weird.

PR #8212 (2018-03-05): Fixes multilane's ToRoadPosition #8045
In this PR `multilane`'s version of `ToRoadPosition` has been modified to take into account:

- `linear_tolerance`: when two different `lanes` whose output `distance`s from `ToLanePosition` with the same `GeoPosition` are within `linear_tolerance`, the code now keeps on analyzing the r-coordinate.
- Even if `distance` is 0, the checks go over all the `lane`s of the `Segment` to check that there isn't any other `lane` with a smaller r-coordinate.

A test case has been created to evaluate it the changes and prior tests were remained (note that there was no  [... cut]
closed 2018-03-05 by agalbachicar