RobotLocomotion/drake #8049: Proximity engine seems to report wrong results with cylinders

amcastro-tri, 2018-02-13:
I created a [simple test](https://github.com/amcastro-tri/drake/blob/proximity_engine_with_cylinders/geometry/test/proximity_engine_test.cc#L460). A vertical cylinder against a half space. In principle, if you rotate the cylinder about its center 180 degrees, you should get the same result. This is not the case, the proximity engine reports **very** different results.
This test prints the values of the computed depths. When there is penetration, the reported depth should be "0.051". At zero degrees rotation that is exactly what you get. With 180 degrees rotation you get "0.001".
Maybe I got the frames wrong?

amcastro-tri, 2018-02-13:
You can see a [video of the simulation of a cylinder falling against a half space](https://www.youtube.com/watch?v=-tR3t73mzC8&feature=youtu.be) (no friction, that is why it only moves up and down and rotates but doesn't move laterally since contact forces only act in the z-direction). The dynamic is performed by `MultibodyPlant` while penetration queries are handled through `GeometrySystem`.
You can see that while the cylinder's z-axis points "up" the simulation looks right with the cylinder bouncing off the plane at its sharp edges. However, as the cylinder starts rotating with its z axis pointing "down", it starts going through the plane. Further analysis would seem to indicate that the upper half of the cylinder would seem to be half the actual size.

avalenzu, 2018-02-13:
[This](https://github.com/flexible-collision-library/fcl/blob/43048336c34a01156dc216e8534ffb2788675ddf/include/fcl/narrowphase/detail/primitive_shape_algorithm/halfspace-inl.h#L362) is the function that actually computes the intersection.

amcastro-tri, 2018-02-13:
ok, [this is the offending line](https://github.com/flexible-collision-library/fcl/blob/43048336c34a01156dc216e8534ffb2788675ddf/include/fcl/narrowphase/detail/primitive_shape_algorithm/halfspace-inl.h#L374) (or one of them). It should say instead:
```c++
// this checks for cylinder axis aligned with plane normal (that's why the math is simpler!)
if(std::abs(cosa) < halfspaceIntersectTolerance<S>()) { ...
  // ^^^^^ missing abs().
```

Another thing I didn't like about this implementation is the very loose tolerance used in the check. `halfspaceIntersectTolerance()` resolves to `1e-7` for `double` precision.
I didn't test it, but the code in [GeometricTools here](https://www.geometrictools.com/GTEngine/Include/Mathematics/GteIntrPlane3Cylinder3.h) seems to look cleaner.

amcastro-tri, 2018-02-14:
Fix pushed in flexible-collision-library/fcl#255

avalenzu, 2018-04-19:
@amcastro-tri can this be closed?

amcastro-tri, 2018-04-23:
Yes. Thanks @avalenzu

closed 2018-04-23 by amcastro-tri