RobotLocomotion/drake #8058: QP Inverse Dynamics costs are not positive semidefinite

rdeits, 2018-02-13:
@hongkai-dai I just tried using your new SCS branch in the QP controller, and I got the same "Y is not positive definite" error. It turns out that error is coming from the factorization in `drake::math::DecomposePSDmatrixIntoXtransposeTimesX`. The actual problem is that some of the quadratic costs in the QP inverse dynamics problem are indeed not positive definite or even positive semidefinite. Specifically, the left and right foot cost matrices have eigenvalues of -0.717458. These are large enough that even if we summed all the cost matrices instead of factorizing them separately the full cost matrix would still not be PSD. This looks like a bug in QP inverse dynamics itself, but I'm not sure what's going on yet.

hongkai-dai, 2018-02-13:
Cool, glad that `DecomposePSDmatrixIntoXtransposeTimesX` captured a real bug. I was worrying that the function name was too long when I wrote it.

rdeits, 2018-02-13:
Never mind--the `-0.7` eigenvalue was due to rounding in *printing* the Y matrix when I was checking its eigenvalues. The actual eigenvalues of that matrix are all > -4e-10. The factorization was failing because ScsSolver uses a default tolerance of -1e-10, which was just a bit too restrictive for this particular problem. 

Unfortunately, changing that tolerance to -1e-9 results in `SCS returns code -6, with message "SCS unbounded inaccurate".` I'll open an updated issue with the actual problem.

closed 2018-02-13 by rdeits