RobotLocomotion/drake #8060: Upload SNOPT 7.5 sources into git for use by CI

jwnimmer-tri, 2018-02-13:
I'd like to finish #1647 by submitting a PR to Drake that incorporates the SNOPT 7.5 sources.  Since Drake CI is run using MIT's SNOPT license, someone who has access to SNOPT 7.5 via MIT (not me via TRI) will first need to upload that 7.5 release to somewhere accessible to CI.  For now I'll tag @jamiesnape as a guess, but maybe it can or must be reassigned.

For the purposes of this issue, I'm looking to end up with a branch or tag under https://github.com/RobotLocomotion/snopt with the unpacked 7.5 source archive, that I would cite via its git commit hash in my Drake PR.

\CC @RussTedrake @EricCousineau-TRI FYI

jamiesnape, 2018-02-14:
+@RussTedrake if you can point me to a source archive, I can do the upload.

RussTedrake, 2018-02-15:
It looks like my old software download links don't work anymore.  I can ask the developers.  But to be clear, it looks like they have versions 7.6 and 9.0 now.  But I'm asking for 7.5?

jamiesnape, 2018-02-15:
That is my understanding, though I would be happy to upload 7.5m 7.6, and any other versions to different branches if you happen to obtain copies of that source

jwnimmer-tri, 2018-02-15:
I think 7.5 would be best for now to match TRI, but maybe 7.6 is similar enough not to matter.  I agree with Jamie that the more the merrier, we can upload them all.

jamiesnape, 2018-02-23:
We have the 2018-01-20 release which includes SNOPT 7.6.1 imported with commit https://github.com/RobotLocomotion/snopt/commit/c17db3769e59d4a8d651631d5d79641cecca0504.

jamiesnape, 2018-02-23:
@RussTedrake you may want to toggle on branch protection for the branch `2018-01-20` in the admin console for that repo.

RussTedrake, 2018-05-17:
aok by me.  I've set up the branch protection (only admins can push).

closed 2018-05-24 by jwnimmer-tri