RobotLocomotion/drake #8061: SCS fails to solve QPInverseDynamics problem

rdeits, 2018-02-13:
Replacing Gurobi with the SCS solver results in two issues. The first is that transforming the costs into rotated Lorenz cone constraints causes `DecomposePSDmatrixIntoXtransposeTimesX` to fail because SCS sets its minimum eigenvalue to `-1e-10`, but the QPInverseDynamics body motion costs are around `-3e-10`. We can fix that by changing the tolerance in `ScsSolver.cc`, but that just results in SCS itself failing with `SCS returns code -6, with message "SCS unbounded inaccurate".` (this is using the new SCS on the `upgrade_scs` branch). 

Vastly decreasing the contact cost weights (from 1e5 to 1e0) gets SCS from `SCS unbounded inaccurate` to `SCS solved inaccurate`, which still causes the controller to fail with `QpInverseDynamicsSystem: QP cannot solve`.

rdeits, 2018-02-13:
Ok, will do!

jwnimmer-tri, 2021-11-11:
We've had OSQP available for a while (and changed ChooseBestSolver a bunch since then).  Can someone re-test to see whether this bug is still present or not?

jwnimmer-tri, 2023-05-22:
Closing for lack of reproducible problem report.  If there are new details that demonstrate a problem, please re-open.

closed 2023-05-22 by jwnimmer-tri