RobotLocomotion/drake #8074: pydrake: AddValueInstantiation doesn't place the class in the correct module

jadecastro, 2018-02-14:


EricCousineau-TRI, 2018-02-16:
Reproduction commit (from virtual f2f (v2v?)):
https://github.com/jadecastro/drake/blob/416123ad47f53556ac36e378e83c9b43de15b750/bindings/pydrake/automotive_py.cc#L39

EricCousineau-TRI, 2018-02-17:
It actually wasn't an issue with the wrong module being used (the template instantiation was being registered correctly, just the wrong *name* was being printed) - rather, the input is somehow getting converted to a `BasicVector<double>`? (at least, that's how RTTI sees it)
https://github.com/EricCousineau-TRI/drake/commit/4edc923221c3f3948fc28813aaf0e13d9306c3dc

The output:
```Yar:  <pydrake.systems.framework.AbstractValue object at 0x7f212bd668f0>
[ 0.  0.  0.  0.]
 OutputSteeringCommand: data type: 1
 stuff: drake::systems::Value<drake::systems::BasicVector<double>>
```

So it sees it correctly in the Diagram context, but once it gets funneled to the system, something goes awry?

EricCousineau-TRI, 2018-02-17:
Error was due to some mishaps I introduced in our pybind fork; luckily, the fix was relatively easy:
https://github.com/RobotLocomotion/pybind11/pull/10

PR #8094 (2018-02-21): framework_py: Add test of using AbstractValue's in input ports, fix overload error for unique_ptr in pybind
Closes #8074

Address part of `AbstractValue` point in #7660. (Just need to follow-up with unrestricted updates.)

When I was first prototyping accepting `unique_ptr` as arguments in `pybind`, I introduced a bug where *any* registered pybind type would match with *any* other pybind type. This is fixed via https://github.com/RobotLocomotion/pybind11/pull/10
(I ultimately made this a possibility when supporting anti-slicing stuff for `shared_ptr`, but just hadn't incorporated it until now. [... cut]
closed 2018-02-21 by EricCousineau-TRI