RobotLocomotion/drake #8090: Unexpected behavior when dynamically changing realtime rate

basicNew, 2018-02-16:
I've been playing a bit with the realtime rate on the simulator via [Simulator::set_target_realtime_rate](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/simulator.h#L135). The simulation worked as I would expect when the rate was set before running the simulation. However, I was a bit surprised when I started to change the rate dynamically, while the simulation was running:

- If I increased the rate (e.g. going from 1 to 2) the sim would run at max speed until it catches up with the desired rate.
- If I decreased the rate (e.g. going from 2 to 1) the sim would pause until it catches up with the desired rate.

This behavior became evident when I reached the [Simulator::PauseIfTooFast](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/simulator.cc) implementation, which considers the time since the simulation started, the current time and the realtime rate (however I found it a bit puzzling from the end-user POV). So, couple of questions / thoughts:

- If this is the expected behavior, maybe it would be worth adding it to the documentation in `set_target_realtime_rate`? I would be happy to send a PR for this.
- I would assume that `PauseIfTooFast` uses the accumulated time (and not compare on each step time) to avoid accumulating small errors and try to be as close as possible to the requested rate. Is that correct?
- I see I can fix this by calling [Simulator::ResetStatistics()](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/simulator.cc#L36) before calling `set_target_realtime_rate`, however that also nukes all the other stats (like e.g. `num_steps_taken_`) which I would like to keep. Is there another way to achieve this (i.e. not loosing all the stats)? If not, do you think it would be worth exploring to add support for this kind of dynamic rate-changing without loosing the stats? Again, would be happy to send a PR for this too.

cc @stonier @clalancette

EricCousineau-TRI, 2018-02-16:
One minor modification to the third option would be to add an option, something like `pause_using_accumulated_clock_time`, which you could set to false for use cases where you'd like a knob to control the realtime rate during the simulation.

@sherm1 What do you think?

jwnimmer-tri, 2018-02-16:
In my opinion, the set_realtime_rate is intended to satisfy the basic, non-interactive use case.  For code that wants to have a time scrubber or DVD-menu like tools, calling `StepTo` and manually managing whatever clocks are necessary seems perfectly fine.  In other words, I don't think the framework needs to consolidate this kind of feature, it just needs to provide sufficient (and clear) APIs to allow downstream code to get the job done.

In other words, my vote is for the first bullet-option.

sherm1, 2018-02-16:
Thanks, Andrés. Can you say what the application is for this on-the-fly speed changing?

sherm1, 2018-02-16:
(So far I agree with @jwnimmer-tri -- but I'd like to understand the motivation.)

basicNew, 2018-02-16:
Sure @sherm1 . The motivation is basically what @jwnimmer-tri describes: having an interactive UI that allows to control different aspects of a simulation, one of those being the real time factor. We originally started by doing the simulation loop ourselves, but after digging a bit more, we realized we would be (almost) replicating the code that handles the real-time factor in the `Simulator` class. So we thought it would be worth to share our experience and ask for your advise on what the preferred way of handling this is.

sherm1, 2018-02-16:
OK, I see. I took a look at the code. I think it would be reasonable to have set_target_realtime_rate() reset initial_simtime_ and initial_realtime_ using the same code that is currently in ResetStatistics(). You could factor out those two lines into a private method and invoke them from either place. Would that make it behave as you would prefer?

If so, I'd be happy to see a PR that would make this change (don't forget unit tests!).

basicNew, 2018-02-16:
That sounds good to me @sherm1 , I'll try to send a PR on Monday.

sherm1, 2018-02-17:
/cc @edrumwri

edrumwri, 2018-02-17:
Since the real time rate is computing an average and since the real time
rate was introduced for interactive applications, why not use an
exponentially weighted moving average? That would seem to be able to solve
the problem in an elegant and simple way.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

sherm1, 2018-02-17:
I like that a moving average would allow the realtime rate to recover automatically from user-induced pauses or breakpoints. Can you sketch an algorithm, @edrumwri? It's not obvious to me how the weight and time constant would be chosen (should they depend on the selected rate?), how the average would be updated for variable steps, and whether it requires a state variable.

edrumwri, 2018-02-17:
Sure- sounds fun. I'll work on it this weekend.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2018-02-19:
As promised, here is an analysis of the problem and sketch of the solution.

Designate cᵥ = v₁ + ... + vₙ as the accumulated virtual time after n steps
in the simulation and cₐ = a₁ + ... + aₙ the accumulated real (wall clock)
time after the same n steps. The actual (i.e., not targeted/requested)
"real time factor" is computed as cᵥ / cₐ.

The problem that Andres identified is that all of these samples (vᵢ and aᵢ)
are equally weighted. This means that if the simulation requires unusually
extensive or little computation  at some sample nearer to 1 than to n, the
change to the real time factor can be greatly affected at n. Intuitively,
we want samples drawn closer to n to have more weight because this is an
interactive application. It should also be clear that it's easy to slow the
simulation down to attain a target real time factor but much harder to
speed it up, because computation limits the speed that virtual time can
progress. The effect is that effecting a target real time factor can be
hard to achieve if the real time factor is suddenly increased or if
simulation steps occasionally require lots of computation.

The solution I propose is to give greater weight to newer samples by
storing a weighted cumulative virtual time and a weighted cumulative real
time. To get a feel for the problem, I considered that we would want a
target real time rate of 1.0 and generated 100 data samples in MATLAB:

>> virtual = rand(100,1)*1e-2 + ones(100,1)     % generate virtual times
tightly clustered around 1.0
>> actual = rand(100,1)*1e-3 + ones(100,1)   % generate actual times more
tightly clustered around 1.0
>> actual(50) = 100;                                        % pretend we
have one simulation step that required much work
>> plot(virtual ./ actual)

We see that the instantaneous (unadjusted) simulation rate stays around
1.0, as expected:

​In other words, if there were no rate adjustment, the simulation would
"hiccup" for a minute and then return to smooth operation. With rate
adjustment, this is what the simulation sees as the real time rate:

>> plot(cumsum(virtual) ./ cumsum(actual))

​In this case (the current implementation), the simulation thinks it must
catch up, which means that it's going to be trying to run faster than real
time to get the rate back to 1.0. Given my problem setup, where real time
and actual time are similar, this means that the simulation would need to
run subtly faster than real time until the average gets back to 1.0.

Now look what happens to the average when we use weighted averages of the
form αⁿ⁻¹v₁ + αⁿ⁻²v₂ + … α⁰vₙ:

>> alpha = 0.5;
>> cum_alpha_virtual(1) = virtual(1);
>> cum_alpha_actual(1) = actual(1);
>> for i=2:100
>> cum_alpha_virtual(i) = cum_alpha_virtual(i-1)*alpha + virtual(i);
>> cum_alpha_actual(i) = cum_alpha_actual(i-1)*alpha + actual(i);
>> end
>> plot

​So the unadjusted real time rate stays much closer to 1.0, effectively
filtering older values.

So, there would need to be a function that computes the update rate (which
would be good because it would allow `get_actual_realtime_rate()` to become
compliant with the styleguide by no longer performing any computation
itself). Possibly correct C++ code follows:

void Simulator<T>::UpdateRealtimeRate() {
  const double virtual_time_now =
ExtractDoubleOrThrow(get_context().get_time());
  const double virtual_time_passed = virtual_time_now - last_virtual_time_;
  last_virtual_time_ = virtual_time_now;
  const auto clock_now = Clock::now();
  const Duration real_time_passed = clock_now - last_real_time_;
  last_real_time_ = clock_now;
  (weighted_virtual_time_ *= realtime_rate_weighting_factor_) +=
      virtual_time_passed;
  (weighted_real_time_ *= realtime_rate_weighting_factor_) +=
      real_time_passed.count();
  real_time_rate_ = weighted_virtual_time / weighted_real_time_;
}

realtime_rate_weighting_factor would be a scalar in [0,1], which I imagine
could be user settable with a good default of 0.5 or so. A setting of 1.0
would effect the default behavior and a setting of 0.0 would effectively
make the real time rate instantaneous (not a good idea because timing is
not generally accurate over small time spans).

Andres, does this algorithm make sense?

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

On Fri, Feb 16, 2018 at 5:35 PM, Evan Drumwright <evan.drumwright@tri.global

> Sure- sounds fun. I'll work on it this weekend.
>
>
> --------------------------------
> Evan Drumwright
> Senior Research Scientist
> http://positronicslab.github.io
> Toyota Research Institute
> Palo Alto, CA
>
>
>> I like that a moving average would allow the realtime rate to recover
>> automatically from user-induced pauses or breakpoints. Can you sketch an
>> algorithm, @edrumwri <https://github.com/edrumwri>? It's not obvious to
>> me how the weight and time constant would be chosen (should they depend on
>> the selected rate?), how the average would be updated for variable steps,
>> and whether it requires a state variable.

sherm1, 2018-02-19:
@edrumwri did you mean to include the plots in your comment?

edrumwri, 2018-02-19:
Argh. It's not letting me do inline plots. It showed up great in my email response :(

basicNew, 2018-02-21:
Thanks a lot @sherm1 and @edrumwri for looking into this! The simple implementation that Michael initially proposed is more than enough for our needs (we are currently using ResetStatistics() to have a first working version). Also, as @jwnimmer-tri pointed out, the requirements from an interactive simulation backend may not be 100% aligned with those of a passive simulation framework, so there is always the case that we may find another requirement in the future that won't be supported by Drake's simulator.  So, we would be supper happy to send a PR with Evan's proposal and contribute to Drake; I just want to make sure this is a valuable addition aside for the original use case presented here, as our requirements may change and we may end up having to implement the clocks on our side anyway.

sherm1, 2018-02-21:
On general principles I would be inclined to implement the simplest fix that satisfies the present use case and wait until we have a clear need before doing a more substantial version that would require more testing. If you do go the simple route please add a TODO mentioning Evan's idea and referencing this issue number.

basicNew, 2018-02-21:
Makes perfect sense, thanks @sherm1 .

basicNew, 2018-05-29:
Finally returning to this issue. A couple of extra requirements came up and, as @jwnimmer-tri suggested, the best thing was to implement the clock handling on the interactive backend itself. 

@edrumwri thanks a lot for the code you drafted, that's what I ended up implementing :).

@EricCousineau-TRI @sherm1 thank you too for your comments in the discussion. 

I'm closing this issue now as I don't think is relevant to Drake itself.

closed 2018-05-29 by basicNew