RobotLocomotion/drake #8093: Make Doxygen High Level Documentation Consistent - Python Bindings, Stochastic Systems

EricCousineau-TRI, 2018-02-16:
Presently, there are two top-level pages which are high-level overviews, `Python Bindings` (which has an awkward `aliases` Module) and `Stochastic Systems`, while there are multiple Modules.

[image]

For consistency, these should be placed under `Modules` with the other high-level documentation.

@RussTedrake Would you be OK with me moving the `Stochastic Systems` page under `Modules`?

\cc @jwnimmer-tri

RussTedrake, 2018-02-18:
Definitely.  Slip it in wherever it fits.  Please note that there are link to some of these pages from sphinx, which may need to be repaired if you make these moves.

EricCousineau-TRI, 2018-10-21:
Resolved via #9494.

closed 2018-10-21 by EricCousineau-TRI