RobotLocomotion/drake #8096: consider generating python bindings in vector_gen

RussTedrake, 2018-02-17:
They are beautiful to have (see #8095), but it's suboptimal to type in the bindings and names manually.  Understand there are a few non-trivial details here, but adding this for our consideration.

jwnimmer-tri, 2018-03-06:
The payoff here is obvious.  I _think_ it would be easy to generate an `*.inc` file during codegen that some other `*_py.cc` module could `#include`, so probably worthwhile.

EricCousineau-TRI, 2018-03-07:
A minor perturbation to that implementation would be to generate a (`*_pybind.cc` + `*_pybind.h`) pair (since these aren't directly consumed by Python), so that way you can have separation compilation units that can be statically linked (thus we put less weight on our already-bloated module bindings :/ ).

It's something I've been wanting to do at some point for the pybind modules.

jwnimmer-tri, 2018-03-07:
@EricCousineau-TRI I think you're best suited to take a pass as the codegen prototype at least (the `//tools/vector_gen:lcm_vector_gen.py` changes that express what the emitted code should look like)?  If you'd like a hand with unit testing it or integrating into `vector_gen.bzl` rules then just let me know.

jwnimmer-tri, 2021-11-12:
Another tactic here would be to add `Serialize` to C++ named vectors, and then to teach pydrake's C++ bindings how to automatically bind a C++ class that has a a `Serialize` function (into Python attributes).

RussTedrake, 2021-11-12:
My own use of vector_gen has all but disappeared, now that we no longer favor using subclasses of `BasicVector` on the ports (nor even in the `State`).  Personally, I would prioritize #12566 over this.

closed 2024-03-21 by jwnimmer-tri