RobotLocomotion/drake #8130: Valkyrie model's left-side OBJ meshes have reversed normals

rdeits, 2018-02-21:
On the left is the .obj file for Val's left arm link 5, and on the right is the .dae file for the same link:

[image]

The meshes for the right side of the robot don't have this problem:

[image]

Simply opening the .dae in Meshlab and and then exporting it as .obj fixes the problem. I'll see if I can do that from the command line.

rdeits, 2018-02-21:
The following script seems to fix the .obj meshes. Note the horrifying need to call `meshlabserver` several times to get it to actually convert the mesh without randomly crashing...

```python
import os
import subprocess

filter_script = os.path.join(os.path.dirname(__file__), "fix_uv.mlx")

for (dirpath, dirnames, filenames) in os.walk(os.curdir):
    print(dirpath)
    for file in filenames:
        if file.endswith(".dae"):
            # meshlabserver crashes frequently and seemingly at random.
            # The horrifying workaround is to just try the same conversion
            # several times.
            for i in range(10):
                try:
                    subprocess.check_call(["meshlabserver", "-i", os.path.join(dirpath, file),
                                           "-s", filter_script,
                                           "-o", os.path.join(dirpath, file.replace(".dae", ".obj")),
                                           "-om", "vt", "vn", "wt"])
                    break
                except:
                    pass
            else:
                throw(ValueError("Couldn't convert mesh: {:s}".format(file)))
```

Question: do we actually *want* these .obj files in Drake? Is it worth updating them if they were apparently somewhat broken all along and no one noticed?

jwnimmer-tri, 2018-03-31:
I think its okay to keep the meshes in Drake.  We are working on a way to mitigate large data files in git, but these in total are only ~9M for both dae and obj, so if we rev the obj files once more it won't make a big difference.

@rdeits Since you have a fix and know better what a correct result looks like, would you like to submit this PR?

PR #8626 (2018-10-23): Re-generate valkyrie obj meshes from dae
Fixes #8130

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8626)
closed 2018-10-23 by jwnimmer-tri