RobotLocomotion/drake #8178: Compiler warning in OSQP

jwnimmer-tri, 2018-02-26:
There is a warning while compiling OSQP, at least under GCC:
```
09:38:51 INFO: From Compiling external/osqp/src/osqp.c:
09:38:51 external/osqp/include/osqp.h: In function 'osqp_solve':
09:38:51 external/osqp/src/osqp.c:410:5: warning: 'can_print' may be used uninitialized in this function [-Wmaybe-uninitialized]
09:38:51    if(!can_print){
09:38:51      ^
09:38:51 external/osqp/src/osqp.c:228:8: note: 'can_print' was declared here
09:38:51   c_int can_print; // Boolean whether you can print
09:38:51         ^
```

If this is a real bug, we should PR or file an issue upstream.

In any case, its undesirable for our build to emit warnings, so for now we should suppress the warning message.  We do that by editing `copts = []` in the `cc_library` rule in `package.BUILD.bazel` such as https://github.com/RobotLocomotion/drake/blob/858390b/tools/workspace/snopt/package.BUILD.bazel#L164.

hongkai-dai, 2018-02-27:
Filed a PR to OSQP upstream at https://github.com/oxfordcontrol/osqp/pull/51

hongkai-dai, 2018-02-27:
The PR has been merged to OSQP upstream. Will send a PR to drake to update the SHA of osqp

hongkai-dai, 2018-02-28:
Resolves by #8205

closed 2018-02-28 by hongkai-dai