RobotLocomotion/drake #8197: Bazel build has a mix of `-iquote` and `-isystem` for external packages

EricCousineau-TRI, 2018-02-27:
Relates #8137 #8146

When building with `bazel build -s //...`, we saw the following:

```
  /usr/bin/clang-4.0 -U_FORTIFY_SOURCE '-D_FORTIFY_SOURCE=1' -fstack-protector -Wall -B/usr/bin -B/usr/bin -fcolor-diagnostics -fno-omit-frame-pointer -g0 -O2 -DNDEBUG -ffunction-sections -fdata-sections '-std=c++1y' -MD -MF bazel-out/k8-opt/bin/external/bullet/_objs/libdrake_bullet_collision.so/external/bullet/src/BulletCollision/CollisionShapes/btTetrahedronShape.pic.d '-frandom-seed=bazel-out/k8-opt/bin/external/bullet/_objs/libdrake_bullet_collision.so/external/bullet/src/BulletCollision/CollisionShapes/btTetrahedronShape.pic.o' -fPIC -DBT_USE_DOUBLE_PRECISION -iquote external/bullet -iquote bazel-out/k8-opt/genfiles/external/bullet -iquote external/bazel_tools -iquote bazel-out/k8-opt/genfiles/external/bazel_tools -isystem external/bullet/src -isystem bazel-out/k8-opt/genfiles/external/bullet/src -isystem external/bazel_tools/tools/cpp/gcc3 -fvisibility-inlines-hidden -Wno-all -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/bullet/src/BulletCollision/CollisionShapes/btTetrahedronShape.cpp -o bazel-out/k8-opt/bin/external/bullet/_objs/libdrake_bullet_collision.so/external/bullet/src/BulletCollision/CollisionShapes/btTetrahedronShape.pic.o)
```

The main thing to note that is there is both `-isystem` and `-iquote` for Bullet headers, and it's unclear when exactly this is happening.

It's not clear if this is how we've configured our libraries, or if this is something with upstream bazel.

\cc @jamiesnape

m-chaturvedi, 2018-05-11:
1. Bazel transitively adds `-isystem` for header files which have been [included with `includes`.](https://docs.bazel.build/versions/master/be/c-cpp.html#cc_library.includes)

2. It transitively adds `-iquotes` for header files which have been [included with `hdrs`.](https://docs.bazel.build/versions/master/be/c-cpp.html#cc_library.hdrs)

3. We get `-isystem external/bullet/src` because we have [`includes = ["src"]`](https://github.com/RobotLocomotion/drake/blob/066518613005a0a4f17d2dc66989d493f2a35105/tools/workspace/bullet/package.BUILD.bazel#L34)

4. We get `-iquote external/bullet` because Bazel adds the current directory with `-iquote`.

This behavior can be reproduced with [simpler examples.](https://github.com/m-chaturvedi/try_bazel), copy-pasted from [here](https://github.com/bazelbuild/examples) 

If we build `with_iquote` with `bazel build -s //main:hello-world` we can see that the current directory is included with an `-iquote`.  `-isystem` is used only for 'real' system includes.  And `#include "lib/hello-time.h"` works fine in `hello-world.cc`, but `#include <lib/hello-time.h>` doesn't work which we expect since [headers included with angular brackets are not searched with -iquote](https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html)

If we build `with_isystem` with `bazel build -s //main:hello-world` we can see that the [only relevant change in the log is that of `-isystem lib`(sorry that gist diff is not specific enough)](https://gist.github.com/m-chaturvedi/9de8874847517090fb43a3c11c010c66/revisions) and we can include `hello-time.h` with either `#include "lib/hello-time.h"` or `#include <lib/hello-time.h>`.

m-chaturvedi, 2018-05-11:
[Relevant discussion](https://groups.google.com/forum/#!topic/bazel-discuss/HyyuuqTxKok)

jwnimmer-tri, 2018-05-11:
@EricCousineau-TRI I don't understand what problem we are trying to solve here?  What is the resulting undesirable outcome that these kinds of spellings incur?

jwnimmer-tri, 2018-05-15:
Assuming we always spell non-toolchain (i.e., non-libc, non-libstdc) includes with quotes, can you give an example where things don't work?  Is the problem with installed Drake, perhaps?

jamiesnape, 2018-05-15:
Installed Drake should be fine. CMake uses `-I` unless you explicitly specify `SYSTEM`, in which case it uses `-isystem`.

EricCousineau-TRI, 2018-05-15:
Aside from the issues we've already ran into and worked around, I do not yet have a concrete example; however, I fear that if we, or a user, have incorporate a third-party library `B` which depends on another library `A` that we include, but it relies on bracket includes, then it may leak in the wrong transitive dependency from the system.

I have not confirmed this yet, though, so I can see if I can whip up a reproduction example.

jwnimmer-tri, 2018-05-15:
I thought that's why we did `includes = ["."]` in many of our externals?

In any case, the possible problem is much clearer now, thanks.  I guess I'll await a demonstration before I measure whether this issue is still valuable to pursue.

m-chaturvedi, 2018-05-15:
[Correct link for angular brackets being used internally](https://github.com/eigenteam/eigen-git-mirror/blob/master/unsupported/Eigen/MatrixFunctions#L17)

m-chaturvedi, 2018-05-17:
After meeting with Eric, I realized I was not clear with my observations.  I try again:

> I thought that's why we did includes = ["."] in many of our externals?

`includes = ["."]` seems to be needed because [at least one library, Eigen, uses angular brackets](https://github.com/eigenteam/eigen-git-mirror/blob/master/unsupported/Eigen/MatrixFunctions#L17) in their code and [`-iquote` only applies to quoted headers](https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html).

> Assuming we always spell non-toolchain (i.e., non-libc, non-libstdc) includes with quotes, can you give
> an example where things don't work? Is the problem with installed Drake, perhaps?

Eigen is [using old-style cast](https://github.com/eigenteam/eigen-git-mirror/blob/master/Eigen/src/Core/util/Macros.h#L922).  When C code is included inside C++ using quoted headers included via `-iquote` clang doesn't suppress the warnings related to old-style casts, [which it does with headers included via `-isystem`](https://clang.llvm.org/docs/UsersManual.html#controlling-diagnostics-in-system-headers).  We have had [similar problems with LCM before](https://github.com/lcm-proj/lcm/pull/222) because in Drake we have `-Werror` for `-Wold-style-cast`.

jwnimmer-tri, 2018-05-17:
Hah, well since I guess I added it #7407, I should be the one to answer that.  I guess I would say that if `Wold-style-cast` is somehow causing problems we can turn it off again, because of cpplint.  But since we have it on right now, I guess I'm still unclear on what problem this issue is trying to resolve.

Or perhaps more succinctly: @EricCousineau-TRI this issue needs a priority assigned.  My intuition is "Priority: Backlog" because I don't see what harm its causing, but if someone has a case for "Priority: High" then I'd like to hear it.  We shouldn't be working on backlog issues.

jwnimmer-tri, 2021-11-11:
I don't think this issue is very useful nor actionable.  If we experience build failures down the road that lead back to here, we can re-open it.

closed 2021-11-11 by jwnimmer-tri