RobotLocomotion/drake #8225: Find resource falling over on relative loaded library path

stonier, 2018-03-02:
For the installed drake binaries, the `LoadedLibraryPath("libdrake.so")` path returns a relative path in some workspace configurations. This will bomb any lookup for objects contained in packages, e.g. prius mesh files.

```
ldd UNIT_automotive_simulator_test
       .....
       libignition-transport4.so.4 => /tmp/delphyne/install/lib/libignition-transport4.so.4 (0x00007f37b3c3e000)
        ../../../install/lib/libdrake.so (0x00007f37b2af6000)
        libignition-msgs1.so.1 => /tmp/delphyne/install/lib/libignition-msgs1.so.1 (0x00007f37b26be000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f37b233c000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f37b2033000)
```

jwnimmer-tri, 2018-03-02:
I can't connect the dots from that report and snippet to exactly what's going wrong.  Is `../../../install/lib/../share/drake` not the correct answer?  Is the problem in fact with `FindResource`, or perhaps with `PackageMap::PopulateUpstreamToDrake`?  Could you pose the error text that ends up coming out?  Could you post more of a reproduction recipe?

jwnimmer-tri, 2018-03-02:
Over to @jamiesnape, at least for delegation.

stonier, 2018-03-04:
**prius_vis.cc**

* `const std::string sdf_filename = FindResourceOrThrow(     "drake/automotive/models/prius/prius_with_lidar.sdf");`
  * this creates a relative path due to the loaded library path being relative
* `parsers::sdf::AddModelInstancesFromSdfFileToWorld(sdf_filename, ...`

**sdf_parser.cc:** AddModelInstancesFromSdfFileToWorld(filename, ...)

* `const string full_path_filename = GetFullPath(filename);`
  * this converts the relative path to an absolute one
  * will be oft broken depending on where you call it since it uses `getcwd`
  * but typically ok because the input path is usually absolute
*  `package_map.PopulateUpstreamToDrake(full_path_filename);`

**package_map.cc** PopulateUpstreamToDrake(model_file)

* `model_file` is the `getcwd` derived absolute path
* `drake_path = MaybeGetDrakePath();`
  * this calls `FindResource("drake/.drake-find_resource-sentinel")`
  * which will also return a relative path as before
* `std::mismatch(drake_path.begin(), drake_path.end(), model_dir.begin());`
  * and boom, since `drake_path` is relative, but `model_dir` is absolute

It'd probably work if you take the problematic `GetFullPath` out, but you might want to visit `FindResource` and decide if that should always return an absolute path or not (the variable names inside that function would seem to indicate that it is expected to be an absolute path).

jwnimmer-tri, 2018-03-05:
That makes sense, thanks.   Indeed since `FindResourceResult::get_absolute_path()` is documented to return the absolute path, the implementation should be doing that, either when collecting candidates, or when publishing the final result.

jwnimmer-tri, 2018-03-12:
From my quick reading either (1) `std::vector<optional<string>> candidate_dirs` always has to be absolute paths and `FileExists` should assert as much, or else (2) `FileExists` can accept relative paths but then convert them to absolute upon return.

stonier, 2018-03-12:
I expect nailing it in (1), more specifically in the methods that contribute to (1), makes more sense. By the time you get to (2) you don't have the context and subsequently it is ambiguous what the path might be relative to (the binary? the pwd?).

jwnimmer-tri, 2018-03-12:
That sounds right.  Also note that `AddResourceSearchPath(std::string root_directory);` hazards the same defect, and should likely be fixed at the same time (or else the assertion would trip).

fbudin69500, 2018-03-12:
The problem with relative path, is that it has to be relative to something. The path to `libdrake.so` can potentially be relative, as @stonier showed, but this it relative to the executable that is currently running, whereas some other paths could be relative to `cwd`. I think that to avoid any ambiguity, only absolute paths should be accepted. The question is how to handle the error: exception? assert? return value?

jwnimmer-tri, 2018-03-12:
> I think that to avoid any ambiguity, only absolute paths should be accepted.

Do you mean for the case of `AddResourceSearchPath`.  I think as long as we document in its API the requirement for absolute paths, the exact nature of the failure message is less important.  I wouldn't use a return value though -- either DRAKE_DEMAND or DRAKE_THROW_UNLESS make more sense.

jwnimmer-tri, 2018-11-09:
Fixed by #8369.

closed 2018-11-09 by jwnimmer-tri