RobotLocomotion/drake #8230: Provide a way to create a Vector/Matrix of symbolic Variable / Expression

soonho-tri, 2018-03-02:
For now, we do have some methods in `MathematicalProgram` but we can factor them out to `common` to serve users better.

@RussTedrake , I'll add my proposal after dinner. Please feel free to add a comment with the signatures that you have in mind.

soonho-tri, 2018-03-03:
@RussTedrake , how is the following names/signatures? I will also add `MakeMatrix*` versions.

```c++
// In `common/symbolic_variable.h` file.

namespace drake {
namespace symbolic {

VectorX<Variable> MakeVectorXVariable(int rows, const std::string& name,
                                      Variable::Type type);
VectorX<Variable> MakeVectorXBooleanVariable(int rows, const std::string& name);
VectorX<Variable> MakeVectorXBinaryVariable(int rows, const std::string& name);
VectorX<Variable> MakeVectorXContinuousVariable(int rows,
                                                const std::string& name);
VectorX<Variable> MakeVectorXIntegerVariable(int rows, const std::string& name);

template <int rows>
Eigen::Matrix<Variable, rows, 1> MakeVectorXVariable(const std::string& name,
                                                     Variable::Type type);
template <int rows>
Eigen::Matrix<Variable, rows, 1> MakeVectorXBooleanVariable(const std::string& name);
template <int rows>
Eigen::Matrix<Variable, rows, 1> MakeVectorXBinaryVariable(const std::string& name);
template <int rows>
Eigen::Matrix<Variable, rows, 1> MakeVectorXContinuousVariable(const std::string& name);
template <int rows>
Eigen::Matrix<Variable, rows, 1> MakeVectorXIntegerVariable(const std::string& name);

}
}

soonho-tri, 2018-03-05:
TODO:
 - [ ] Add matrix versions.
 - [ ] Clean up `MathematicalProgram` to remove duplicates and use the ones introduced by this.

jwnimmer-tri, 2021-11-18:
>  Add SymmatricMatrix versions (feature-request @hongkai-dai)

Let's punt this to a new issue, if we even still need it.

> Clean up MathematicalProgram to remove duplicates and use the ones introduced by this.

That's the remaining call to action here.

jwnimmer-tri, 2024-03-16:
We can revisit any cleanups here next time some specific code is being modified and runs into trouble.

closed 2024-03-16 by jwnimmer-tri