RobotLocomotion/drake #8231: homebrew python3 issues

soonho-tri, 2018-03-02:
#8228 fixes some problems but we still have other issues. I've only tested `bindings/pydrake` but I found that the following tests failed:

```
//bindings/pydrake/util:_isolated/cpp_param_pybind_test                  FAILED in 0.5s
//bindings/pydrake/util:_isolated/cpp_template_pybind_test               FAILED in 0.3s
//bindings/pydrake/util:cpp_param_pybind_test                            FAILED in 0.5s
//bindings/pydrake/util:cpp_template_pybind_test                         FAILED in 0.3s
```

Here is a snippet of an error message:

```
==================== Test output for //bindings/pydrake/util:_isolated/cpp_template_pybind_test:
[ RUN      ] CppTemplateTest.TemplateClass
unknown file: Failure
C++ exception with description "ImportError: dlopen(/usr/local/opt/python2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_ctypes.so, 2): Symbol not found: __PySlice_AdjustIndices
  Referenced from: /usr/local/opt/python2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_ctypes.so
  Expected in: flat namespace
 in /usr/local/opt/python2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_ctypes.so
```

jamiesnape, 2018-03-05:
`export PATH=/usr/local/opt/python@2/libexec/bin:/usr/local/opt/python@2/bin` appears to necessary, though not necessarily sufficient. I also had to remove and reinstall various `pip` dependencies.

jamiesnape, 2018-03-06:
@jwnimmer-tri @EricCousineau-TRI How would you feel about switching any executable Python files to using `#! /usr/bin/env python2.7`?

jwnimmer-tri, 2018-03-06:
We currently have a mix of:
```
#!/usr/bin/env directorPython
#!/usr/bin/env python
#!/usr/bin/env python2
#!/usr/bin/python
```

Given that the only real debate is `/usr/bin/env` (use `$PATH`) vs `/usr/bin/python` (locked down), and our http://drake.mit.edu/code_style_guide.html#additional-rules says to use the shebang, using a better shebang seems eminently reasonable to me.

Though if you have any links to where it was debated or objected to in the past, I'd be curious to see that context.

Since its more important to have this correct now given homebrew, seems worth adding a linter guard to it -- likely within the current drakelint'er would be easy.

jamiesnape, 2018-03-06:
The easiest fix for the linter would be to just whitelist `/bin/bash`, `/bin/sh`, `/usr/bin/env python2.7` for shebangs, or I could focus on the various combinations of `python` and `bash` and let anything go. Any preference?

jwnimmer-tri, 2018-03-06:
I would be okay with a hard-coded whitelist.

EricCousineau-TRI, 2018-03-07:
Yup, that all sounds good to me!

soonho-tri, 2018-03-10:
https://github.com/RobotLocomotion/drake/blob/28e5d1060e8bffa084e9db7fdabc4ef6ebd1ff7f/setup/mac/source_distribution/install_prereqs_user_environment.sh#L11-L23

On my mac, I don't have `/usr/local/opt/python@2/libexec/bin/python` (maybe after re-installing `python@2`?). Instead, I have `/usr/local/opt/python@2/bin/python`. Do we need to update this script? (I can open a PR when it's confirmed)

jamiesnape, 2018-03-10:
Did they update the formula again? Seems to have been at least once a day lately. I will probably PR a change to use `python2.7` in the next day or two, but everything is so unstable, it is difficult to know what to do for the best.

soonho-tri, 2018-03-12:
I think we are in good shape after those fixes. I've run `bazel test //... -c dbg --config=everything` and there are only vtk-8 related test failures which should be addressed by #8325.

Nice thing is that I don't need to change the `PATH` variable to make `python` to invoke `python2.7`.

PR #8340 (2018-03-12): Follow PEP 394 now that Homebrew (supposedly) does
Closes #8231.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8340)
closed 2018-03-12 by soonho-tri