RobotLocomotion/drake #8253: When SNOPT is enabled, `--copt -g` doesn't work to add symbols to release builds

jwnimmer-tri, 2018-03-05:
When SNOPT is enabled, `--copt -g` doesn't work to add symbols to release builds (for targets that depend on snopt), because of the `-s` flag.  This is a bug due to my PR #7965.  Jamie fixed macOS in #8133 but I guess Ubuntu wasn't immediately as simple.  This is a tracking issue to make sure Ubuntu gets fixed up also.

jamiesnape, 2018-03-05:
Basically the un-versioned symlinks to versioned shared libraries (in practice only VTK) are being installed as duplicates of the target library rather than symlinks. `strip` then chokes when it tries to strip the un-versioned library since (I think) there is a mismatch between the name and the internal metadata. A little strange that it affects `strip`, but something that needs fixing regardless.

jwnimmer-tri, 2018-03-22:
I believe that #8341 has resolved this?

jamiesnape, 2018-03-22:
Yes.

closed 2018-03-22 by jamiesnape