RobotLocomotion/drake #8271: Lint XML files

jamiesnape, 2018-03-06:
At least for well-formedness. Bonus points for validation against relative schemas, if appropriate.

SeanCurtis-TRI, 2018-03-09:
FTR I was playing with xmllint today (as referenced in #8297). I found it struggled with the sdf and urdf schema.

Specifically, the `maxOccurs` and `minOccurs` fields were erroneously flagged for having values that were not 0 or 1. It should support any non-negative number *and* the string "unbounded". So, before we can validate against relevant schema, this will have to be worked out.

This was the behavior I observed on xenial with the apt-get'd `libxml2-utils`.

jamiesnape, 2018-03-11:
The alternative is going to be something the `lxml` python module, especially as we already have it in our prerequisites.

avalenzu, 2018-03-21:
This just came up again in #8410. That PR fixes a missing `<` at the start of an `origin` tag.

jamiesnape, 2018-03-21:
Yes, that would require validation against the schema, as it was technically well-formed. I will bump the priority.

SeanCurtis-TRI, 2018-03-21:
Bad XML syntax we can do now. And that would've caught the error that @avalenzu pointed out. But I still haven't seen a solution that'll let us validate against the schema.

jwnimmer-tri, 2018-03-26:
Note: Probably we should lint `*.xacro` files the same as `*.xml` files, at least by default.

jamiesnape, 2018-03-26:
Yes, I have `*.config`, `*.sdf`, `*.world`, `*.urdf`, `*.xacro`, and `*.xml` on my list so far.

EricCousineau-TRI, 2020-07-29:
This came up as a pain point in #13758... `xmllint` is already on our system (or at least mine, from `libxml2-utils`) and looks good.

Here's an example of applying to all of the XML files I know of: #13768

EricCousineau-TRI, 2020-07-30:
FTR I didn't see any occurrence of the issues that Sean mentioned here: https://github.com/RobotLocomotion/drake/issues/8271#issuecomment-371948320

SeanCurtis-TRI, 2020-07-30:
Yep. I'd attribute that to #13326.