RobotLocomotion/drake #8316: cpplint cannot (necessarily) count correctly on Mac

jamiesnape, 2018-03-09:
```
'/// d𝐱/dt = f(t, 𝐱; 𝐤), where f : t ⨯ 𝐱 → ℝⁿ, t ∈ ℝ, 𝐱 ∈ ℝⁿ, 𝐤 ∈ ℝᵐ, provided an'
```

Ubuntu: 80 characters
Mac with Python 2: 86 characters
Mac with Python 3: 80 characters

Problems appear to be with `𝐱` and `𝐤` being counted as two characters in the above.

fbudin69500, 2018-03-09:
Related to https://github.com/RobotLocomotion/drake/pull/8318

m-chaturvedi, 2018-03-22:
https://github.com/google/styleguide/issues/337

jwnimmer-tri, 2018-06-26:
If wanted to merge up to the latest tip of upstream `google/styleguide` in order to obtain this fix, that would mean we'd intake a many new `cppguide.html` changes, which would take the platform review team a few weeks to chew through.

Would everyone be okay if instead we only merge the google/styleguide#337 commit into `RobotLocomotion/styleguide` in a first PR, and then follow-up with the full merge later?  My understanding is that because we are using (only) merge commits now, it should be painless (modulo any conflicts in the `cpplint.py` itself, which should be easily solved / solveable).

What to you think @EricCousineau-TRI @jamiesnape @m-chaturvedi @ggould-tri etc.?

jamiesnape, 2018-06-26:
Sounds good to me.

m-chaturvedi, 2018-06-27:
Sure!

jwnimmer-tri, 2018-07-10:
@m-chaturvedi if you'd like to propose this PR to `RobotLocotion/styleguide`, I could review it.  Personally, I would be fine either with pulling only the 337 change, or else pulling _all_ of the commits that touched `cpplint.py`.

commit 2ae3f6f6 (2018-07-11): Update SHA for styleguide. Closes #8316.
commit 525c2958 (2018-07-11): Merge pull request #9110 from m-chaturvedi/update_styleguide

Update SHA for styleguide. Closes #8316.
closed 2018-07-11 by m-chaturvedi