RobotLocomotion/drake #8322: drake-visualizer crashes on Ubuntu 16.04 with NVIDIA drivers

avalenzu, 2018-03-09:
I see the following behavior on my Ubuntu 16.04 desktop (GeForce GTX 1080 Ti 11GB):
```
$ sudo ./setup/ubuntu/16.04/install_prereqs.sh
$ bazel run --config everything //:install ~/.local
$ ~/.local/bin/drake-visualizer
ERROR: In /vtk/Rendering/OpenGL2/vtkShaderProgram.cxx, line 475
vtkShaderProgram (0x4763d60): Shader object was not initialized, cannot attach it.
$ bazel run //tools:drake_visualizer
ERROR: In /vtk/Rendering/OpenGL2/vtkShaderProgram.cxx, line 475
vtkShaderProgram (0x4763d60): Shader object was not initialized, cannot attach it.
```

avalenzu, 2018-03-09:
I see this error under `nvidia-384` and `nvidia-390`. With nouveau there is no problem.

EricCousineau-TRI, 2018-03-09:
If I do `git revert -m 1 f6bada0` (reverting #7851, using old VTK), then VTK runs fine.

If we don't have a fix by EOD, would it be possible to temporarily revert this?
(I have some things on hardware I'd like to test, and would like to stay close to `master`.)

BTW, @jamiesnape I've assigned you for now - is that OK?

jamiesnape, 2018-03-09:
We had some people not able to run the previous version, and CI has been upgraded, so it is non-trivial.

EricCousineau-TRI, 2018-03-09:
Though it may not have caught this on CI, I'll see if I can whip up a quick test to actually run `drake_visualizer` on CI and finally address #7050

jamiesnape, 2018-03-09:
Trying a make VTK 8.0 version of Director. Linux is building now, but Director seems to be broken due to Homebrew Python issues.

EricCousineau-TRI, 2018-03-09:
Just following up from the Slack convo, I tested `rgbd_camera_test` on `master`, and it failed with the following errors:
```
$ git log -n 1 --oneline
383a5aa Bind PoseBundle, FrameVelocity, PoseAggregator, SpatialVector, SpatialVelocity (#8282)
$ bazel run :rgbd_camera_test 
...
[==========] Running 7 tests from 3 test cases.
[----------] Global test environment set-up.
[----------] 1 test from RgbdCamera
[ RUN      ] RgbdCamera.TestInstantiation
ERROR: In /vtk/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx, line 785
vtkXOpenGLRenderWindow (0x563e1a54aaf0): GLEW could not be initialized.

ERROR: In /vtk/Rendering/OpenGL2/vtkXOpenGLRenderWindow.cxx, line 753
vtkXOpenGLRenderWindow (0x563e1a54aaf0): failed to create offscreen window

ERROR: In /vtk/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx, line 785
vtkXOpenGLRenderWindow (0x563e1a54aaf0): GLEW could not be initialized.
```

EricCousineau-TRI, 2018-03-13:
BTW, I believe at some point on Friday, @avalenzu tried using the revert commit, and had an installed version of Drake leaking in its version of `vtk-8.1` which that was shadowing Bazel's version, even when building within Bazel.

Andres, can I ask if you were able to resolve this by uninstalling Drake from `~/.local`, and installing it to something like `~/.local/opt/drake`?

jwnimmer-tri, 2018-03-13:
> ... installed version of Drake leaking in its version of vtk-8.1 which that was shadowing Bazel's version, even when building within Bazel.

That should (separately?) be fixed, at least on Ubuntu.  There's no reason to allow a user's bash LD_LIBRARY_PATH nor PYTHONPATH to make it past our drake-visualizer launch script.

airfield20, 2018-03-13:
I also had this issue, doing the git revert command above fixed the issue.

jamiesnape, 2018-03-13:
> That should (separately?) be fixed, at least on Ubuntu. There's no reason to allow a user's bash LD_LIBRARY_PATH nor PYTHONPATH to make it past our drake-visualizer launch script.

I can fix it here.

PR #8789 (2018-05-16): Update to VTK 8.1 and latest Director and add packages for Bionic
Needs local testing on Linux of `drake-visualizer` using both `bazel run //tools:drake_visualizer` and the installed executable (`bazel run install -- /opt/drake && /opt/drake/bin/drake-visualizer`). Note that for licensing reasons this switches to using a shared library for LCM (c.f., #7423, #8153).

Fixes #7908, fixes #8322, fixes #8764. Relates #7907.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8789)
closed 2018-05-16 by sammy-tri