RobotLocomotion/drake #8323: Process to deprecate old functionality in rotation_matrix_deprecated.h

mitiguy, 2018-03-09:
In the Drake codebase, there are multiple call sites to old/soon-to-be-deprecated functions.  These old functions currently exist in the file drake/math/rotation_matrix_deprecated.h.  

The sequence of events to deprecate and then remove these old functions (and to eventually delete the file rotation_matrix_deprecated.h) is as follows:

1.  Replace all calls to old functions in rotation_matrix_deprecated.h with calls to their newer equivalents in the RotationMatrix class (the RotationMatrix class is in the file drake/math/rotation_matrix.h).

Note: If the call sites to an old function have limited scope/occurrence within the Drake codebase, then all calls may be eliminated in a single PR.  If there are multiple calls in multiple parts of the Drake codebase, there may be multiple PRs to avoid a "flag day".

2.  Concomitant with the PR that eliminates the last call in the Drake codebase to an old function,
the file rotation_matrix_deprecated.h will be edited.
a.  A TODO message will be added, with text similar to the following:
    // TODO(mitiguy) This code was deprecated on March 9, 2018.
    // Delete this code in accordance with issue #ThisIssueNumber.

b.  Replace the body of the old function in rotation_matrix_deprecated.h with something like:
  DRAKE_DEPRECATED("Function is deprecated.  Use RotationMatrix::NameOfMethod().")

3.  Within a reasonable amount of time (3 months), and ideally before the next major release (e.g., prior to release 1.0), delete the TODO, the DRAKE_DEPRECATED message and all vestiges of the old function.

4.  As soon as all old functions have been deleted, also delete the file rotation_matrix_deprecated.h and its #include from drake/math/rotation_matrix.h

EricCousineau-TRI, 2018-03-09:
Can I ask what the plan is for this deprecation w.r.t. `RigidBodyTree`?
Will `RigidBodyTree` be part of the deprecation, in line with having `MultibodyTree` in its stead (once all equivalent functionality is provided)?
Or will `RigidBodyTree` be migrated to the newer utilities?

sherm1, 2018-03-09:
Per discussion with @mitiguy, this issue is narrowly focused on just the deprecated math routines, and no functionality gets lost. Deprecating RBTree (which we definitely want to do!) is a much bigger issue and will have to wait until MBTree can take over. Where RBTree uses these deprecated methods, its call sites will get fixed but RBTree will live for a while longer.

EricCousineau-TRI, 2018-03-09:
Gotcha! So internally, it would use the newer methods / data types, but the API itself would remain the same?

sherm1, 2018-03-10:
Right!

jwnimmer-tri, 2018-12-21:
I believe that #9183 has resolved this issue.  Please re-open with details if not.

closed 2018-12-21 by jwnimmer-tri