RobotLocomotion/drake #8329: clang-format collides with linter on a static_cast function pointer

RussTedrake, 2018-03-10:
this happens often in `mathematicalprogram_py.cc`.  the linter wants this:
```
static_cast<VectorXIndeterminate(MathematicalProgram::*)(
               int, const std::string&)>(
               &MathematicalProgram::NewIndeterminates),
           py::arg("rows"), py::arg("name") = "x")
```
but clang-format undoes it to this:
```
static_cast<VectorXIndeterminate (MathematicalProgram::*)(
               int, const std::string&)>(
               &MathematicalProgram::NewIndeterminates),
           py::arg("rows"), py::arg("name") = "x")
```
resulting in errors like:
```
bindings/pydrake/solvers/mathematicalprogram_py.cc:173:  Extra space before ( in function call  [whitespace/parens] [4]
```

jwnimmer-tri, 2018-03-16:
The `static_cast` spelling is dis-preferred by http://drake.mit.edu/doxygen_cxx/python_bindings.html in the "Function Overloads" section.  After changing it to the suggested `py::overload_cast` instead, the lint error switches to to an `operator<` whitespace false positive, which we can suppress with a `NOLINT`:
```patch
   prog_cls.def(py::init<>())
       .def("NewContinuousVariables",
-           static_cast<VectorXDecisionVariable(MathematicalProgram::*)(
-               int, const std::string&)>(
+           // NOLINTNEXTLINE(whitespace/operators)
+           py::overload_cast<int, const std::string&)>(
                &MathematicalProgram::NewContinuousVariables),
```

I'll kick the bug back to you @RussTedrake in case you'd like to keep it open in order to assign someone to fix up the spellings, or you can close now that there's a suggestion if you prefer.

RussTedrake, 2018-03-17:
did you try compiling your version?  Neither py::overload_cast nor overload_cast_explicit seem to work in these cases.

jwnimmer-tri, 2018-03-17:
I thought I had?  But you're right, that doesn't compile, so I must have done it wrong.

The answer that _does_ compile is similar though:
```patch
--- a/mathematicalprogram_py.cc	2018-03-17 08:21:09.613864952 -0400
+++ b/mathematicalprogram_py.cc	2018-03-17 08:20:21.985543945 -0400
@@ -191,16 +191,19 @@
   py::class_<MathematicalProgram> prog_cls(m, "MathematicalProgram");
   prog_cls.def(py::init<>())
       .def("NewContinuousVariables",
+           // NOLINTNEXTLINE(whitespace/parens)
            static_cast<VectorXDecisionVariable (MathematicalProgram::*)(
                int, const std::string&)>(
                &MathematicalProgram::NewContinuousVariables),
            py::arg("rows"), py::arg("name") = "x")
       .def("NewContinuousVariables",
+           // NOLINTNEXTLINE(whitespace/parens)
            static_cast<MatrixXDecisionVariable (MathematicalProgram::*)(
                int, int, const std::string&)>(
                &MathematicalProgram::NewContinuousVariables),
            py::arg("rows"), py::arg("cols"), py::arg("name") = "x")
       .def("NewBinaryVariables",
+           // NOLINTNEXTLINE(whitespace/parens)
            static_cast<VectorXDecisionVariable (MathematicalProgram::*)(
                int, const std::string&)>(
                &MathematicalProgram::NewBinaryVariables),
@@ -231,11 +234,13 @@
                    MathematicalProgram::*)(const Variables&, int)>(
                &MathematicalProgram::NewSosPolynomial))
       .def("NewIndeterminates",
+           // NOLINTNEXTLINE(whitespace/parens)
            static_cast<VectorXIndeterminate (MathematicalProgram::*)(
                int, const std::string&)>(
                &MathematicalProgram::NewIndeterminates),
            py::arg("rows"), py::arg("name") = "x")
       .def("NewIndeterminates",
+           // NOLINTNEXTLINE(whitespace/parens)
            static_cast<MatrixXIndeterminate (MathematicalProgram::*)(
                int, int, const std::string&)>(
                &MathematicalProgram::NewIndeterminates),
```

The overall solution that lets us get rid of `// NOLINT` comments is #4843 -- to mandate clang-format and then disable cpplint's formatting-related checks.

PR #8394 (2018-03-19): add nolint lines and run clang-format on mathematicalprogram_py.  
no code changes (only whitespace/comments).  
resolves #8329 (for now)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8394)
closed 2018-03-19 by jwnimmer-tri