RobotLocomotion/drake #8331: PoseAggregator Python binding trying to move an InputPortDescriptor

sherm1, 2018-03-10:
After a merge with master I am now getting this CI failure in my caching branch:
```
ERROR: test_pose_aggregator (rendering_test.TestRendering)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./bindings/pydrake/systems/test/rendering_test.py", line 136, in test_pose_aggregator
    "pose_and_velocity", instance_id2)
RuntimeError: return_value_policy = move, but the object is neither movable nor copyable!
```
This binding was recently added in PR #8282, and is wrapping this method of PoseAggregator:
```c++
  std::pair<const InputPortDescriptor<T>&, const InputPortDescriptor<T>&>
  AddSinglePoseAndVelocityInput(const std::string& name, int model_instance_id);
```
That method creates a couple of input ports and returns references to them. I don't think it should be trying to move the InputPortDescriptors!

The reason this fails in the caching branch is that I removed the ability to move InputPortDescriptors, which is generally a bad thing to do.

@EricCousineau-TRI, @jadecastro does this actually make sense here?

jadecastro, 2018-03-11:
@sherm1, I think this might be a simple matter of telling pybind to return references because it unfortunately defaults to move semantics unless you tell it otherwise.

I have this branch that fixes it:
https://github.com/RobotLocomotion/drake/compare/master...jadecastro:pybind-fix-move

(Depending on what you prefer, I could either PR the change now, or else you could subsume the change in yours.)

sherm1, 2018-03-11:
Thanks, @jadecastro! I tried your fix in my branch but it had no effect -- same error message. Likely that is because of the bizarre return type of that method: `std::pair<const InputPortDescriptor<T>&, const InputPortDescriptor<T>&>`. The pair itself is not a reference -- it's being returned by value. But its _contents_ are references. I'm guessing that the `py_reference_internal` refers to the return argument so probably ought still to be a value return in this case.

One option might be not to wrap this particular method -- is it needed?

jadecastro, 2018-03-12:
I agree. What I believe is that, sadly, this is a limitation of `pybind11`'s reference semantics in the context of `std::pair`. Do you agree, @EricCousineau-TRI?

I would like to keep the binding for `AddSinglePoseAndVelocityInput` if at all possible. I will try coming up with a solution that works with your branch.

jwnimmer-tri, 2018-03-12:
I think maybe you need to wrap the method using a custom function in the bindings layer, instead of the default template voodoo, and then in your function manually unpack the pair into a pythonic object?

jadecastro, 2018-03-12:
Here's a quick proposal for a fix: https://github.com/RobotLocomotion/drake/commit/a6fac428b8ed7f08f4a3b5621b8f753ac976786e

It's a little more extreme than what Jeremy suggests in the sense that I suggest that `PoseAggregator` return a container rather than a pair. It requires a little more verbosity at the call sites, but removes the `first`/`second` semantics, and is somewhat easier for pybind to digest. Let me know if this is a sensible route.

sherm1, 2018-03-12:
@jadecastro I interpret the comments so far as an endorsement of your idea to provide a custom return container for AddSinglePoseAndVelocityInput(), with the caveat that it can probably be done with just a minimalist struct.

(P.S. I hate GitHub's feeble code discussion tools! Much easier in Reviewable.)

EricCousineau-TRI, 2018-03-12:
I think I'd be more of a fan of @jwnimmer-tri's suggestion. Something like:
```
// Allow RVP to be specified for a tuple.
template <typename... Ts, int... Is>
py::object CastTuple(auto tup, vector<py::return_value_policy> rvps) {
  DRAKE_DEMAND(rvps.size() == sizeof...(Ts));
  return py::make_tuple(py::cast(std::get<Is>(tup), rvps[I])...);
}

// In binding:
cls.def("returns_tuple", []() { return CastTuple(returns_tuple(), {py_reference, py_reference}); });
```

EricCousineau-TRI, 2018-03-12:
Er, another option would be to convert the lvalue references to pointers, which would also work:
```
cls.def("returns_tuple", []() { auto tup = returns_ref_tuple(); return std::make_pair(&tup.first, &tup.second); });
```

(sorry for mixing `tuple` and `pair`; I'm not sure if `std::get` works with `std::pair`.)

PR #8347 (2018-03-13): PoseAggregator's AddSinglePoseAndVelocityInput() returns a struct
Resolves #8331.

Resurrecting #8343 (since we can no longer re-open that due to force-push)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8347)
closed 2018-03-13 by jadecastro