RobotLocomotion/drake #8336: Custom URDF not loading : packagepath error

Sreevis, 2018-03-11:
I tried to load drake example description externally with absolute path. For example, I copied kinova description outside drake directory and successfully loaded using its absolute path towards URDF.

But when I tried the same with my own URDF description package it fails to load and showing this.
`[warning] Couldn't find package 'robot_description' in the supplied packagepath.`
`[warning] Mesh 'package://robot_description/meshes/base_link.STL' could not be resolved and will be ignored by Drake. If you don't want it to be ignored, please include it in the package map.
terminate called after throwing an instance of 'std::runtime_error'`
`what():  external/drake/multibody/parsers/urdf_parser.cc: ParseGeometry: ERROR: Mesh file name could not be resolved from the provided uri "package://robot_description/meshes/base_link.STL".
Aborted (core dumped)`

hongkai-dai, 2018-03-14:
Do you use ROS? Could you try to load the URDF from ROS? If ROS cannot load your URDF either, then it's a problem in your URDF.

Do you have a `package.xml` file in the your folder? One example is in https://github.com/RobotLocomotion/drake/blob/master/examples/valkyrie/package.xml. With this `package.xml` file, the parser knows how to interpret `package://valkyrie` in the URDF file https://github.com/RobotLocomotion/drake/blob/master/examples/valkyrie/urdf/urdf/valkyrie_A_sim_drake_one_neck_dof_wide_ankle_rom.urdf#L15

Sreevis, 2018-03-14:
Thank u.
Now I am using drake alone.
As u said, it was an error in my urdf package. When I put the relative path for meshes it worked.

closed 2018-03-14 by Sreevis