RobotLocomotion/drake #8352: Consider supporting Python 3

EricCousineau-TRI, 2018-03-13:
I believe we've discussed this before, but I'm not sure if it's been captured in a Drake-specific issue. Just writing this down for now, so I've marked it as `priority: low`. It can be upped if need be.

Options:
* Support both Python3 and Python2.7 (for ROS)
* Support only Python3

Pros:
* Infix `@` for matrix multiplication NumPy
* Just being up-to-date (e.g. Homebrew pivots would hopefully hurt us less?)
* [Any other specific libraries?]

Cons:
* If supporting 2+3:
  * Shebang inflexibility? (we just had to switch to `/usr/bin/env python2` from `/usr/bin/env python2.7` from `/usr/bin/env python`)
  * More compatibility shims
  * More CI configurations
* If only supporting 3:
   * Loss of ROS support for Python? (@clalancette May I ask what you know what the plan is for distros beyond Lunar?)

Requires:
* `pybind11` support - `pybind11` itself support Python3; our fork (and related features) supports Python3 because we use the same CI: RobotLocomotion/pybind11#11 tests - https://travis-ci.org/RobotLocomotion/pybind11/builds/345471530
* Build switches

\cc @RussTedrake

jwnimmer-tri, 2018-03-13:
I would be surprised if we could get away with a python3 flag day, so I suspect that any plan would involve supporting both for a while (at least in our published APIs), even if eventually we drop python2.

stonier, 2018-03-13:
Recommend writing the above:

```
Support both Python3 and Python2.7 (for System/ROS)
```

The primary reason to keep support for python 2.7 is because that is Xenial's default. Anyone working on the assumption that the system default is a good baseline for development across many packages will want this (ROS just happens to be such a user).

This of course can change when we switch to Bionic beaver (python 3 only), but a consideration would be whether we support the tax of supporting both in the interim.

stonier, 2018-03-13:
> Loss of ROS support for Python? (@clalancette May I ask what you know what the plan is for distros beyond Lunar?)

ROS officially supports whatever the system default is actively works on having their core libraries python3 ready (i.e. 2.7 and 3 compatible) and publically encourages others to do so as well.

From [REP 3](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021)

> **Python**

> Our intent with Python is to track the minimum version provided in the supported Ubuntu platforms, as well as survey other commonly used OS platforms that support ROS to determine a reasonable minimum target.

> Melodic

> Ubuntu will most likely stop supporting Python 2 in release 20.04. To make sure ROS will be able to support that version of Ubuntu, ROS Python packages starting with Melodic Morenia are highly encouraged to support both Python 2.7 and Python 3.5 or later. During the development of Melodic there will be work undertaken to support both Python 2 and Python 3 (including rosdep keys) so ROS package developers can more easily test with either version of Python.

jwnimmer-tri, 2018-03-13:
Good points.  For Ubuntu, I think it would make sense to keep Drake's Xenial support at python2 only, and only support python3 as of Bionic.  We'd still need the source tree to support both, but only certain runtime configurations would be tested with CI.

jamiesnape, 2018-03-13:
Given everything that happened with Homebrew and Python 2/3, this is going to be tricky (and not necessarily dependent on whether we support Python 3 for Mac).

rdeits, 2018-03-16:
My personal feeling is that supporting python3 is extremely important, given that python2 will be completely out of support in ~1.5 years. Drake and director are the last two pieces of software I ever use that don't support python3 yet. 

Fortunately, supporting python2 and 3 simultaneously is much easier than it used to be (difficulty supporting both versions was one of the problems in early releases of python3). The `six` module includes some useful compatibility shims for functions like `range()` and `dict.items()` that have changed their behaviors.

rdeits, 2018-03-16:
Also, problems with homebrew and system python conflicts could be *completely eliminated* by using python's virtual environments (which are even built-in on python3.5 and above). I can't recommend that highly enough.

EricCousineau-TRI, 2018-04-11:
@kevinleestone mentioned that his team is interested in transitioning to Python3 once the surrounding ecosystem is ready, and the teams should strive for Python3 compatible code. He pointed towards Tensorflow as an excellent example of a Python2+3 compatible library, which has the following style guide considerations: https://www.tensorflow.org/community/style_guide#python_2_and_3_compatible

EricCousineau-TRI, 2018-04-15:
Quickly spike-tested what it'd take to support both Python2+3. At the `pybind`, it's generally straightforward if we only support one version of Python per build.
I did a quick hack to `python/repository.bzl` to auto-detect the Python version, warning if it's an unsupported version. I got `symbolic_test` and `autodiffutils_test` (without experimental NumPy features) and a few other things to pass with minor changes.

Some items that would pave the way:
- [ ] Ensure upstream C extension libraries offer `lib/python{version}` artifacts for python 2.7 (both) + 3.5 (Ubuntu 16.04) + 3.6 (Mac Homebrew), e.g. `lcm`, `vtk`, `drake_visualizer`
- [ ] Include whatever python3 dependencies in `install_prereqs.sh` (`six` included)
- [ ] Upgrade codebase using `six` where appropriate

My guess is it'd be a man week of PEH to fully get it working, and more on top of that to ensure we have flavors of CI covering Python3.

Left off here:
[branch](https://github.com/EricCousineau-TRI/drake/tree/feature/py3-wip) | [commit](https://github.com/EricCousineau-TRI/drake/commit/97e8927631eea82774eb1abd40203d69eba6d951)

Still no timeline on this, as I still have it at a relatively low priority. I will see if I can periodically make PRs which slowly make compatibility changes.

EricCousineau-TRI, 2018-07-22:
An additional use case library-wise: `multiprocessing`s forks and `cv2`s threads don't play well together:
https://github.com/opencv/opencv/issues/5150#issuecomment-199318160

One of the workarounds is to use the `spawn` method for `multiprocessing`, which is not available in Python2. Also, from looking at some related issues, it seems like Python3 fixes a few other issues in this library.

pvarin, 2018-09-13:
Just wanted to +1 for Python3 support. I'm starting to use some libraries that are Python3 only (specifically OpenAI baselines) which is making development pretty difficult.

jamiesnape, 2018-09-13:
Python 3 is next on our TODO list of low priority items after Ubuntu Bionic support, so it will certainly happen in the medium term.

EricCousineau-TRI, 2018-10-06:
I am starting to focus on this to see what the challenges may be; this still takes a backseat to Bionic support in general.

That being said, we may want to make our story straight with Bionic + Python2 support, e.g. say that Python2 is experimental, and once Python3 comes online, will be the only supported version.
@jwnimmer-tri @jamiesnape Can I ask y'all's thoughts on this? Would that be too abrupt of a change for users?

RussTedrake, 2018-10-06:
I think as far as the MIT class is concerned... just need the docker instance (locked on 16.04) to continue to work through the fall with Python 2.

jamiesnape, 2018-10-08:
FWIW Bionic support is marked as "experimental" as a whole (not that "experimental" really means a great deal).

> @jamiesnape Can I ask how we should start handling Python3 dependencies?

Dependencies are easy enough. I will fix them once we are in a reasonable situation with the Python docs.

jamiesnape, 2018-10-08:
Also, what is the reasoning for the change to medium priority? This was not even the highest low priority item a couple of days ago. This requires a lot of work on CI and distribution infrastructure that isn't going to happen overnight.

EricCousineau-TRI, 2018-10-08:
> Also, what is the reasoning for the change to medium priority? [...]

Given Jeremy's above point, I would like to see if we can move to Python3-only on upcoming platforms (Bionic, Mojave as well), hopefully without any sort of flag day (e.g. before going from experimental to supported), and save on support matrix combinatorics (e.g. trying to do Python2+3 on Xenial+Bionic+Mojave+etc.).

I apologize that I hadn't acted on this sooner to incorporate in our prior plans.

> [...] This requires a lot of work on CI and distribution infrastructure that isn't going to happen overnight.

Understood that it wouldn't happen overnight; however, I am interested in simplifying our support, if possible. I would be more than up for helping move it forward if we think it's still feasible to do in conjunction with official Bionic support; however, if it is infeasible, I can certainly place the priority back down.

> I would also add, that not sure I see much benefit in pure build scripts running in Python 2 on one platform and Python 3 on another. [...]

If Bazel provided a better mechanism to run both Python2 and Python3 scripts out of the box as native targets, I'd be up for doing build scripts in Python2, and then consumables in Python2 or Python3.
However, I'm a bit hesitant seeing what was necessary to run `mkdoc.py` (even if it were minimal).

If you think it'd be trivial otherwise, I'd be fine with it, but to me, it seems easiest just to support one Python version, period, on a platform (at least to start)?

jamiesnape, 2018-10-08:
> If Bazel provided a better mechanism to run both Python2 and Python3 scripts out of the box as native targets, I'd be up for doing build scripts in Python2, and then consumables in Python2 or Python3.
However, I'm a bit hesitant seeing what was necessary to run mkdoc.py (even if it were minimal).

I am talking things that rely on shebang lines.

EricCousineau-TRI, 2018-10-16:
PRs in flight:
- [ ] Initial update for `pydrake` to be py3 compatible, explicit dep on `six` (#9613)
- [ ] Enable supporting multiple Python versions in Drake's bazel setup (#9649)
- [ ] Initial update for build tools to be py3 compatible (#9615)
- [ ] Miscellaneous updates (#9617)

PRs to submit:
- [ ] Work with shebang discrepancies
    * Anything with non-trivial dependencies will run on same version as Bazel. Anything with trivial dependencies (which are easily available on all systems) can run on either Py2 or Py3.
- [ ] Add bionic Python3 dependencies

If these land in time, we can consider how to handle the support matrix for Bionic + Py2/Py3. All else, we support both on Bionic?

EricCousineau-TRI, 2018-12-22:
Jamie has added some experimental jobs:
https://drake-jenkins.csail.mit.edu/view/Python%203/

They are currently failing due to `deprecation_test` (at least in the one I saw), so I'll need to fix those before nightlies are enabled, so it'll be after the holidays when we get this landed. Getting close!

EricCousineau-TRI, 2019-01-10:
Closed by checkbox-ness.

closed 2019-01-10 by EricCousineau-TRI