RobotLocomotion/drake #8362: Sugar for decltype(T() < T()) as drake::formula_over_t<T>

jwnimmer-tri, 2018-03-14:
In various PRs / issues / slack we've talked a few times about having sugar for a T-dependent `bool` type to replace the `decltype(T() < T())` that we use now.  Since the `decltype(...)` is starting to appear in more Doxygen'd APIs, its probably time to add it?

Possibly we could name it `drake::formula_t<T>` `drake::formula_over_t<T>` or similar.

What do you think @soonho-tri?

sherm1, 2018-03-14:
How about `drake::bool_t<T>`? That would make these cases readable -- I don't think `drake::formula_t<T>` does so.

soonho-tri, 2018-03-14:
`bool_t<T>` might give an impression that its value is either True or False while `bool_t<Expression>` which is `Formula` can hold more than that.

amcastro-tri, 2018-03-14:
+1 for `bool_t<Expression>`. Nicely documented it'd make a nice clickable link in Doxygen docs of methods returning this (vs. the `#if` crap we are using right now).

jwnimmer-tri, 2018-03-14:
I agree with Soonho.  `bool_t` is misleading, because this value is not a boolean, its a formula whose variables are `T`-typed.  We can use its Doxygen to explain that for double and Autodiff its `bool`.

Though I wonder... if it can be Formula sometimes, then the user should never use it in `if`, only ever in `ASSERT` or `cond`.  Perhaps `formula_t<double>` should not be `bool`, but rather a bool-wrapper with ExtractBoolOrThrow, and magic to work with `ASSERT` and `cond`?

jwnimmer-tri, 2018-03-14:
... and then if indeed it has static checking to never screw up (a wrapper in the case of `double`), I could see my way to splitting the difference: `bool_t` as the typedef name, but since it has type checking it wouldn't be accidentally misused.

sherm1, 2018-03-14:
That sounds great. We need a way to capture that the return type is intended to be a bool. Even if under the covers it is something more general, it still has to have true/false semantics. IMO it would be better to have a variety of template sugar classes conveying intent even if they all ultimately have to be formula underneath, rather than lumping them all into a `formula_t`.

soonho-tri, 2018-03-15:
How about the followings?

 - Use `bool_t<T>` which provides implicit bool conversion if a return value should be either `True` or `False` regardless of a given scalar-type `T`. In case of `T = double` or `T = AutoDiff...`, it should work as if we have a `bool`. In case of `T = symbolic::Expression`, the explicit bool conversion for `symbolic::Formula` is triggered when a `bool_t<symbolic::Expression>` is constructed and throws an exception if we can't evaluate a formed `symbolic::formula` such as `x < y`.

 - Use `formula_t<T>` for a general case (i.e. you do not expect a Boolean value in case of `T = symbolic::Expression`).

> Though I wonder... if it can be Formula sometimes, then the user should never use it in if, only ever in ASSERT or cond.

@jwnimmer-tri , FYI/FWIW, `if(v)` is no different from `if(bool{v})` ([source](http://en.cppreference.com/w/cpp/language/implicit_conversion#Contextual_conversions)). If we want to disallow `if(v : formula_t)` then we should not provide a bool conversion for it(neither implicit nor explicit). I am afraid that this will make things less usable.

mitiguy, 2018-03-16:
I am not sure how this is best implemented 
(ideally in a simple-to-use way, even if it has a less simple implementation).
Something with "bool" in the name could be helpful.

jwnimmer-tri, 2018-03-16:
> @jwnimmer-tri , FYI/FWIW, if(v) is no different from if(bool{v}) (source). If we want to disallow if(v : formula_t) then we should not provide a bool conversion for it(neither implicit nor explicit). I am afraid that this will make things less usable.

That's the whole point, though.  If you are doing comparisons on T's, you have only two choices: operate on formulas (returning them, joining them, cond, etc), or DRAKE_ASSERT (or etc.).  If you ever turn it into a `bool` or do an `if`, then your hair is on fire and :explosions:.  We should _not_ allow users to easily demote from formula to bool, since it do not want to encourage code that fails when people start to use Variables.  My vote is for a helper with semantics equal to your `formula_t`, and also add ExtractBoolOrThrow to _explicitly_ ask to break the glass (analogous to ExtractDoubleOrThrow -- you break it you buy it).  If someone wanted your `bool_t<T>`, that's just `bool`, not a template (the extraction happens earlier, not as part of function return signature).  So my vote is still `bool_t<T>` as the name, which (1) is terse, (2) gets the gist across, but (3) has lots of documentation in case someone needs to understand these subtleties.

I also think that Formula should _not_ have even `explicit bool`; that's what ExtractBoolOrThrow is for.

soonho-tri, 2018-03-16:
OK. I'll add:
 - `bool_t<T>`
 - `ExtractBoolOrThrow`

> I also think that Formula should not have even explicit bool; that's what ExtractBoolOrThrow is for.

I added it to make Eigen compatible/compilable with `symbolic::Expression`. I'll check if the version of the Eigen that we are currently using still requires this.

soonho-tri, 2018-03-16:
I have a WIP commit at https://github.com/soonho-tri/drake/commit/a6439f08a9125018ae6a00fd3136307d52c56fec . Any comments are welcome (i.e. name of class/file/target, etc).

soonho-tri, 2018-03-16:
> magic to work with ASSERT and cond

I'll add them.

sherm1, 2018-03-16:
Maybe consider assignment from bool to bool_t<T>? In the witness function code you fixed I saw this:
```c++
bool_t<T> should_trigger(const T& w0, const T& wf) const {
  WitnessFunctionDirection dtype = get_dir_type();
  switch (dtype) {
    case WitnessFunctionDirection::kNone:
      return (T(0) > T(0));
    // ...
   }
}
```
would be nice if that could have been `return false;` (I'm assuming there is such thing as a boolean constant in an Expression.)

amcastro-tri, 2018-03-20:
I am starting to use `Boll<T>` in my unit tests that before returned `decltype(T() < T())`.
It'd seem that I need to say `EXPECT_TRUE(ExtractBoolOrThrow(condition))` for something returning `Bool<T>` while before I was able to compile just fine with `EXPECT_TRUE(condition)`.
@soonho-tri told me there is magic sugar for `DRAKE_ASSERT` and `DRAKE_DEMAND` to work in this way, but not for the GTest macros. Will there be any sugar like that in the future or for now I should just go ahead and place a `ExtractBoolOrThrow` everywhere where I'm expecting a bool (when `T = double`)

soonho-tri, 2018-03-20:
I think for `T = double / AutoDiffXd` cases, one can write `EXPECT_TRUE(condition.value())` which is (slightly) shorter than `EXPECT_TRUE(ExtractBoolOrThrow(condition))`.

For `DRAKE_ASSERT`/`DRAKE_DEMAND` cases, the magic sugar was to specialize `ConditionTraits`. I'm not sure that GTest has a similar mechanism with `EXPECT_*` macros. @jwnimmer-tri , do you have suggestions?

soonho-tri, 2018-03-21:
BTW, while using `Bool<T>`, I found cases where I need logical operators -- `&&`, `||`, `!` -- over `Bool<T>`. I'll open a PR shortly.

jwnimmer-tri, 2018-03-21:
Yeah, I was wondering if we'd need that.  That would be an argument for making `bool_t<T>` a plain typedef, iff we can remove operator-bool conversion from Formula.

soonho-tri, 2018-03-21:
I'll first provide the logical operators for `Bool<T>` and will do a major cleanup over `Bool<T>` when #8417 resolved.

jwnimmer-tri, 2018-03-21:
I would have no objection to sugaring `EXPECT_TRUE(some_Bool)` to automatically call `.value()`, in order for double and Autodiff work easily.

_Probably_ its also okay to sugar it for T = Expression to assume no unbound variables, since test code should usually be fairly certain what's going on with constants vs Variables.

soonho-tri, 2018-03-21:
@jwnimmer-tri , I have the following sugar:

```c++
#include <gtest/gtest.h>

template <typename T>
class Bool {
 public:
 ...
  operator ::testing::AssertionResult() const {
    return ::testing::AssertionResult(value_);
  }
 ...
};
```

which uses the fact that gtest's internal macro [GTEST_TEST_BOOLEAN_](https://github.com/google/googletest/blob/dccc2d67547a5a3a97e4f211f39df931c6fbd5d5/googletest/include/gtest/internal/gtest-internal.h#L1223-L1233) takes `::testing::AssertionResult` in addition to `bool`.

Do you think this is a OK-ish sugar/hack? My concern is that `drake_bool.h` is going to depend on `gtest`.

jwnimmer-tri, 2018-03-21:
Right, you can't put that into `drake_bool.h` because prod code is not allowed to `#include` test code.

soonho-tri, 2018-03-21:
Hmm... another workaround would be to introduce `DRAKE_EXPECT_TRUE` and its variants. But I'm not sure we really want them.

jwnimmer-tri, 2018-03-21:
If gtest doesn't offer a seam that we can use to sugar this down to `EXPECT_TRUE(foo);` only, then I agree that `EXPECT_TRUE(foo.value());` is already good enough.

amcastro-tri, 2018-03-22:
SGTM. I was just passing down my experience with it so you guys had that feedback. I guess in TYPED tests the only option would be to use ExtractBoolOrThrow(), which also SGTM.

soonho-tri, 2018-03-22:
> I guess in TYPED tests the only option would be to use ExtractBoolOrThrow(), which also SGTM.

Since `symbolic::Formula` is bool-convertible for now (while we have an issue to disable it). `.value()` trick should work for now.

Anyway, I don't see a way to make the sugar without violating the "No Include Gtest" rule. I'm closing this issue for now.

PR #8408 (2018-03-20): Support Bool<T> as conditionals in if_then_else and cond
Close #8362 

Note that this PR also moves `:drake_bool` target from `:essential` to `:common` since it depends on `cond` and `autodiff` which are in `:common`.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8408)
closed 2018-03-22 by soonho-tri