RobotLocomotion/drake #8368: DrakeVisualizer publishes a NaN pose at rgbd_camera_publish_lcm_test

kunimatsu-tri, 2018-03-14:
This is how to reproduce:
```
$ cd systems/sensors
$ bazel run :rgbd_camera_publish_lcm_test
```

and this is the result I got from `drake-lcm-spy`:
[image]

Relates: #8345

mitiguy, 2018-03-23:
Overview: The test rgbd_camera_publish_lcm_test is defective.
It has been disabled.  If you have objections, let me know.

History: There is a new rigorous RotationMatrix class.
The functions rotmat2quat and quat2rotmat are now deprecated.

Problem 1:
Due to a defect in the test rgbd_camera_publish_lcm_test, 
there were problems when rotmat2quat was deprecated.  
It was possible to isolate this problem to a single call to rotmat2quat on
line 69 of the file /drake/multibody/rigid_body_plant/viewer_draw_translator.cc.
The following code was initially added to isolate this problem.
    // TODO(mitiguy) when issue #8368 is resolved, change the next line so it
    // instead uses the method drake::math::RotationMatrix::ToQuaternion().
    // Suppress deprecation warning for rotmat2quatOld.
    #pragma GCC diagnostic push
    #pragma GCC diagnostic ignored "-Wdeprecated-declarations"
    auto quat = drake::math::rotmat2quatOld(transform.linear());
    #pragma GCC diagnostic pop  // pop -Wdeprecated-declarations

Problem 2: 
However, when quat2rotmat was also deprecated, the test
rgbd_camera_publish_lcm_test again failed CI.  
Unfortunately, this failure was not isolatable 
to a single call of quat2rotmat.  

Helpful detective hint:
One consistent indication of the defect 
(in both rotmat2quat and quat2rotmat) 
is that elements of a quaternion 
are being assigned NANs.

Note: The following comment now appears in appropriate BUILD.bazel file.
   # TODO(mitiguy) Issue #8368: Test rgbd_camera_publish_lcm_test in the
    # BUILD.bazel file in directory drake/systems/sensors cannot be run anymore
    # due to the new RotationMatrix class which insists on being a proper
    # orthogonal rotation matrix.  This test produces a quaternion with NANs
    # which causes the RotationMatrix class to throw an exception in debug
    # builds. This test is labeled with tags = "manual" so CI does not run it.
    # This test is defective and should never have passed earlier code review.

sherm1, 2018-03-24:
Seems like a good test -- any hope for fixing it soon, @kunimatsu-tri ?

jwnimmer-tri, 2018-03-24:
For the record:
- Relates #8412.
- Use `-c dbg` to reproduce this.  It is a Debug-only check that's failing.

kunimatsu-tri, 2018-03-26:
I'm working on other urgent things right now. Once I finish them, I'll work on solving this hopefully sometime this week.

kunimatsu-tri, 2018-06-09:
It turned out that the bug was coming from wrong SDF modes which don't have `inertia` property.

[image]

PR #8962 (2018-06-11): Add inertia to models
Fixes #8368.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8962)
closed 2018-06-11 by kunimatsu-tri