RobotLocomotion/drake #8374: Question about Drake and ROS

BryceeeeeChen, 2018-03-15:
Hi, I am planning to try out Drake. As a "heavy user" of ROS, I would like to know the reason you guys decided to stop the integration with ROS (I remember you guys did try to make Drake as a catkin package last year). Thanks!

sammy-tri, 2018-03-15:
We've restarted our work on ROS integration, though we're coming from a slightly different direction this time.  https://github.com/osrf/kumonoito contains the current implementation, though it's very work-in-progress at this time., and unfortunately hasn't had a lot of outside testing yet.  If you decide to try it out, any feedback would be fantastic.

naveenoid, 2018-03-16:
It is currently stalled simply because we dont have an immediate need / users but its not left the radar.

BryceeeeeChen, 2018-03-16:
@sammy-tri @naveenoid , thank you for the information. I will look up kumonoito and see if I can modify the interface a little for being used on my own robot. Thanks.

closed 2018-03-16 by BryceeeeeChen

stonier, 2018-03-16:
Aside from @kumonoito, we've also taken the philosophy of just making it compatible for a Xenial and/or ROS ecosystem. That means you can do a `find_package(drake)` and are free to bend it to your own use case as you wish.