RobotLocomotion/drake #8377: `this rule is missing dependency declarations...` error in coverage builds

BetsyMcPhail, 2018-03-15:
`ERROR: /home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/_bazel_ubuntu/d3d6be48f350ff9f6e7359acf59b6b1d/external/sdformat/BUILD.bazel:129:1: Couldn't build file external/sdformat/_objs/sdformat/external/sdformat/src/Filesystem.pic.o: undeclared inclusion(s) in rule '@sdformat//:sdformat':  this rule is missing dependency declarations for the following files included by 'external/sdformat/src/Filesystem.cc': '/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/_bazel_ubuntu/d3d6be48f350ff9f6e7359acf59b6b1d/external/sdformat/include/sdf/system_util.hh'`

For example: https://drake-jenkins.csail.mit.edu/view/Coverage/job/linux-xenial-clang-bazel-experimental-coverage/17/console

It looks like this error is from using **--strict_system_includes** (see #8223).

BetsyMcPhail, 2018-03-15:
Any ideas @jwnimmer-tri?

jwnimmer-tri, 2018-03-16:
Has this always been happening, or did it start failing after previously working?  If so, what changed?

BetsyMcPhail, 2018-03-16:
The error showed up on the nightly job of March 7th - the day that PR was introduced. So the flag + kcov were never working.

(There was a bug in the Bazel-processing code in CDash that caused the error to be ignored. That was fixed yesterday, so the CDash history is incorrect).

jwnimmer-tri, 2018-03-16:
The issue appears to be that the upstream source archive contains a file `include/sdf/system_util.hh` but we are also trying to generate that same file in our `package.BUILD.bazel`, and with `--spawn_strategy=standalone` things get confused.

jwnimmer-tri, 2018-03-16:
Once we are on Bazel 0.11.0, this patch fixes the issue (Bazel 0.10.1 doesn't have `patch_cmds`):
```patch
--- a/tools/workspace/bitbucket.bzl
+++ b/tools/workspace/bitbucket.bzl
@@ -1,5 +1,10 @@
 # -*- python -*-
 
+load(
+    "@bazel_tools//tools/build_defs/repo:http.bzl",
+    "http_archive",
+)
+
 def bitbucket_archive(
         name,
         repository = None,
@@ -56,18 +61,10 @@ def bitbucket_archive(
     if len(repository_split) != 2:
         fail("The repository= must be formatted as 'organization/project'")
 
-    if build_file == None:
-        native.http_archive(
-            name = name,
-            urls = urls,
-            sha256 = sha256,
-            strip_prefix = strip_prefix,
-            **kwargs)
-    else:
-        native.new_http_archive(
-            name = name,
-            urls = urls,
-            sha256 = sha256,
-            build_file = build_file,
-            strip_prefix = strip_prefix,
-            **kwargs)
+    http_archive(
+        name = name,
+        urls = urls,
+        sha256 = sha256,
+        strip_prefix = strip_prefix,
+        build_file = build_file,
+        **kwargs)
--- a/tools/workspace/sdformat/repository.bzl
+++ b/tools/workspace/sdformat/repository.bzl
@@ -12,5 +12,6 @@ def sdformat_repository(
         sha256 = "212211eddd9fa010b4b61a2dae87cd84a66a8b78ed302612d214b7388f9bc198",  # noqa
         strip_prefix = "osrf-sdformat-bac3dfb42cc7",
         build_file = "@drake//tools/workspace/sdformat:package.BUILD.bazel",
+        patch_cmds = ["rm -f include/sdf/system_util.hh"],
         mirrors = mirrors,
     )
```
This has the unfortunate side-effect of printing some extra debug info from Bazel.

Alternatively, we could change `bitbucket_archive` into a `repository_rule()` and add `setup_bitbucket_archive` and then we could `rm` the file after that.

But really, I suspect the best solution is to fix `sdformat` per the TODO in our `package.BUILD.bazel`, and then we don't need to overwrite upstream code in our build rules.

jwnimmer-tri, 2018-03-16:
Back over to @jamiesnape for delegation.

jamiesnape, 2018-03-16:
I think fixing upstream per https://github.com/RobotLocomotion/drake/blob/70405614ade432f1ee59a3a9a90de0d318ff6cbd/tools/workspace/sdformat/package.BUILD.bazel#L86-L93 is definitely the best solution. FYI @clalancette @nkoenig.

BetsyMcPhail, 2018-03-26:
Upstream PR: https://bitbucket.org/osrf/sdformat/pull-requests/421/add-support-for-sdformat_static_define/diff

BetsyMcPhail, 2018-03-26:
Drake PR: https://github.com/RobotLocomotion/drake/pull/8458