RobotLocomotion/drake #8381: clion "bazel build formatter" disagrees with buildifier lint

RussTedrake, 2018-03-16:
I'm constantly getting linter complains about the order here:
```
--- a/bindings/pydrake/BUILD.bazel
+++ b/bindings/pydrake/BUILD.bazel
@@ -172,8 +172,8 @@
         ":test/all_install_test.py",
         ":test/common_install_test.py",
     ],
-    py_dest = PACKAGE_INFO.py_dest,
     targets = PY_LIBRARIES + [":all_py"],
+    py_dest = PACKAGE_INFO.py_dest,
     visibility = ["//visibility:public"],
     deps = get_drake_py_installs(PY_LIBRARIES_WITH_INSTALL),
 )
```

RussTedrake, 2018-03-16:
to be clear, in clion, the build formatter tool automatically re-orders these includes every time I touch the file.  then the command line linter complains and has me run buildifier to switch it back.

jwnimmer-tri, 2018-03-16:
I suspect it is because for our linter and command-line runs we use custom attribute ordering https://github.com/RobotLocomotion/drake/blob/master/tools/lint/buildifier-tables.json#L65, but CLion's built-in copy of `buildifier` doesn't know about that.

Filed upstream as bazelbuild/intellij#261.  In the meantime, I think CLion users could turn off auto-format for these in the CLion menus.

closed 2018-03-16 by jwnimmer-tri