RobotLocomotion/drake #8392: Consider using virtualenv for managing upstream Python dependencies

EricCousineau-TRI, 2018-03-16:
Per @rdeits's comment in #8352:

> Also, problems with homebrew and system python conflicts could be completely eliminated by using python's virtual environments (which are even built-in on python3.5 and above). I can't recommend that highly enough.

My 2 cents:

`virtualenv` does sound like a good option.

I'm not sure if I'm a fan of our present re-inventing of `pip` in Bazel; understandably, it's for reproducibility and keeping most of the dependencies in Bazel, but seems like it may be brittle when the wheels are more complex or have more comprehensive dependencies (e.g. `numpy`). We can always teach Bazel to scan and incorporate the upstream Python dependencies for hermetic-ness. 

Additionally, if the custom-dtype solution for #8116 ends up working, it would greatly smooth out having an optionally patched version of `numpy` for memory management.

That being said, `virtualenv` is additional tweaking on the environment that does have a bit of a funny (and semi-invasive) smell to it. However, if it reduces headaches (e.g. if we can figure out how to teach `nlopt` from Homebrew to not screw around with / conflict with `pip` packgaes), I'm all for it.

@jwnimmer-tri @jamiesnape Can I ask what your opinions are?

jamiesnape, 2018-03-16:
I am pretty much on record with hating reinventing `apt` and `pip` with Bazel, so it would make me happy.

EricCousineau-TRI, 2018-03-16:
(For myself) A tiny breadcrumb for patching local PIP packages:
https://stackoverflow.com/questions/5570666/patching-python-packages-installed-as-dependencies-with-pip

jwnimmer-tri, 2018-03-16:
On the subject of needing `numpy` fixes (not necessarily virtualenv or python2/3 stuff), I am not sure that I'm following, but my main concern is that in the same say `libdrake.so` needs to play well with the rest of the system libraries at runtime, because Drake is a library not a program -- `pydrake` needs to play well with the rest of the python environment at runtime.  If we need to use non-default versions of things like `numpy`, how to we know that other code's use of `numpy` still works?  I think we should go out of our way to be compatible with stable versions of common libraries, and not to require bleeding-edge versions.

jamiesnape, 2018-03-16:
> (For myself) A tiny breadcrumb for patching local PIP packages:
https://stackoverflow.com/questions/5570666/patching-python-packages-installed-as-dependencies-with-pip

I am certainly not in favor of patching packages.

EricCousineau-TRI, 2018-03-20:
Posted responses about package patching in #8116.

jwnimmer-tri, 2018-04-11:
I am not generally as worried about making our development and test environment work.  I am most worried about how the installed pydrake interacts with the rest of the python universe.  I think the clearest way to look at this question (and the wider python2/3 and forked-dependencies questions) is to first explain what the installed experience looks like, and then from there work out what the development environment looks like.

jamiesnape, 2018-04-12:
Given that I am strongly against patching packages, I am not sure a `virtualenv` solves anything in the grand scheme of third-party Drake usage.

EricCousineau-TRI, 2020-09-17:
Ran into another case of desiring this: Better `ipywidgets` support in #14082

jwnimmer-tri, 2023-01-18:
A motivating use case is on the rise -- Homebrew seems incapable of making a Python minor version transition without breaking all of non-first-party Python libraries as they go (e.g., recently `scipy` only works with 3.10, but the default `python3` still points to 3.9).  It's probably that Drake on macOS should _only_ use the Python interpreter from homebreak, and for other python ecosystems dependencies we use `pip`.

jwnimmer-tri, 2024-07-25:
Update: I have somewhat of a prototype of this now in TRI Anzu.  After we gain some experience with that over the next week or two, I'll dump a snapshot of the code into this ticket, along with some more specific goals, and we can take it from there.

jwnimmer-tri, 2024-08-01:
For kitware:
- The first milestone here should be to use a `venv` on macOS.
- We can defer any Ubuntu changes until a later milestone.
- The `venv` will (eventually) need to be part of the provisioned images.

Here's the sample Anzu code to get started: [anzu-snippets.zip](https://github.com/user-attachments/files/16462939/anzu-snippets.zip).

Notes:
- Probably `requirements.{in,txt}` should live in the `setup/...` tree somewhere, not at the project root.
- Also needs changes to `tools/workspace/default.bzl` to call the new `venv_repository` rule.
- Use the venv as `deps = ["@venv"]` on a `py_library` or `py_binary` target.

mwoehlke-kitware, 2024-08-14:
I have no idea what an "anzu" is. I'm guessing all instances of that should be replaced with "drake"? If I do that, I can run the scripts, but it isn't clear how this is supposed to be actually consumed by other Drake bits that need the venv.

Also, if this is intended to replace what's currently `pip install`ed, does that mean we'd drop the `--[without-]test-only` split?

jwnimmer-tri, 2024-08-15:
> I have no idea what an "anzu" is

It's the codename for TRI's internal git repository.

> I'm guessing all instances of that should be replaced with "drake"?

Yes.

> If I do that, I can run the scripts, but it isn't clear how this is supposed to be actually consumed by other Drake bits that need the venv.

That's this part:
> > Also needs changes to `tools/workspace/default.bzl` to call the new `venv_repository` rule.
> > Use the venv as `deps = ["@venv"]` on a `py_library` or `py_binary` target.

> Does that mean we'd drop the --[without-]test-only split?

Probably not.  Users who are installing Drake from source (e.g. via CMake) should only by default need to pay the download cost of the non-testonly dependencies.  The `--with-test-only` should (now) be opt-in, for use only by the Drake Developers (and Drake CI).

jwnimmer-tri, 2024-08-15:
One update from f2f: the unique challenge of "default" vs "testonly" (versus what Anzu does) is to have `sync` know which choice should be used.  The way we can do that is have `install_prereqs.sh` (really, probably `venv/setup`?) symlink the desired `requirements.txt` into or nearby the venv.  Then `sync` can refer to the symlinked file, instead of hard-coding a specific `requirements.txt`.

mwoehlke-kitware, 2024-08-22:
> Use the venv as `deps = ["@venv"]` on a `py_library` or `py_binary` target.

I'm testing this on a system that's had Homebrew freshly reinstalled with nothing installed via `pip`. Pybind11 choked first:
```
ERROR: .../drake/tools/workspace/pybind11/BUILD.bazel:99:37: Action tools/workspace/pybind11/test/sample_header_documentation.h failed: (Exit 1): mkdoc failed: error executing Action command (from target //tools/workspace/pybind11:generate_sample_header) bazel-out/darwin_arm64-opt-exec-ST-a828a81199fe/bin/tools/workspace/pybind11/mkdoc -DDRAKE_COMMON_SYMBOLIC_EXPRESSION_DETAIL_HEADER -DEIGEN_MPL2_ONLY -DSPDLOG_SHARED_LIB -DSPDLOG_COMPILED_LIB ... (remaining 70 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
Traceback (most recent call last):
  File ".../bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 53, in <module>
    from tools.workspace.pybind11.mkdoc_comment import process_comment
  File ".../bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc_comment.py", line 41, in <module>
    from doc.doxygen_cxx.system_doxygen import process_doxygen_to_sphinx
  File ".../bin/tools/workspace/pybind11/mkdoc.runfiles/drake/doc/doxygen_cxx/system_doxygen.py", line 23, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'
```

I tried this:
```diff
diff --git a/tools/workspace/pybind11/BUILD.bazel b/tools/workspace/pybind11/BUILD.bazel
index d94f843278..5dc8d0bcd7 100644
--- a/tools/workspace/pybind11/BUILD.bazel
+++ b/tools/workspace/pybind11/BUILD.bazel
@@ -42,11 +42,12 @@ drake_py_library(
 drake_py_binary(
     name = "mkdoc",
     srcs = ["mkdoc.py"],
     visibility = ["//visibility:public"],
     deps = [
         ":libclang_setup_py",
         ":mkdoc_comment_py",
         "@clang_cindex_python3_internal//:clang",
+        "@venv",
     ],
 )
 
```
...but I get:

```
ERROR: Analysis of target '//bindings/pydrake/geometry:install_lint_genquery_geometry_install' failed; build aborted: no such package '@@venv//': The repository '@@venv' could not be resolved: Repository '@@venv' is not defined
```

Should it be `@venv:something`?

jwnimmer-tri, 2024-08-22:
(Resolved f2f.)

mwoehlke-kitware, 2024-08-26:
Another issue... since we're only doing macOS for now, just adding `deps=["@venv"]` breaks Linux. I suppose there's a way to say "this is only a dep on macOS", but that's a bunch of extra work that we expect to eventually remove. Is there instead a way to make `@venv` a stub (for now) on not-macOS?

jwnimmer-tri, 2025-01-28:
As of today, on macOS we always use a venv (for binary packages, install from source, developers, wheel builds, etc.).

What remains here is to possibly / probably port our Linux builds (i.e., Ubuntu) to also use a venv (or at least offer the option).  However, that is not urgent so I'm lowering the priority here.

jwnimmer-tri, 2026-02-06:
Abandoning in lieu of #22998.

closed 2026-02-06 by jwnimmer-tri