RobotLocomotion/drake #8402: Change `SolutionResult` from enum to enum class, and rename it

hongkai-dai, 2018-03-19:
We will have `SolverResult` soon, it is better to rename `SolutionResult` to avoid ambiguity. I think `SolutionStatus` can be a good name.

avalenzu, 2019-01-28:
`SolutionStatus` sounds good to me. Although I'm always partial to `info` ;-)

jwnimmer-tri, 2019-01-28:
By my reading, most all uses in the API of SolutionResult are to deal with the legacy deprecated APIs -- the MathematicalProgram and MathematicalProgramSolverInterface return value for the non-const Solve method(s).  (There are also a few uses in BranchAndBound, but I'll ignore those for now.)

I would prefer to leave all of that code undisturbed, since we plan to deprecate and then remove it soon, anyway.

The only other use is in MathematicalProgramResult, for `{get/set}_solution_result`.  That's the use we should focus on.

Looking at how calling-code uses this result, almost all call sites are _only_ checking against kSuccess, not using the enumeration details at all.  They are too verbose right now.  We should just offer `bool MathematicalProgramResult::is_success() const` helper for this purpose, so users can easily DEMAND or EXPECT_TRUE or etc on that.

The second most common use is for printing an error message.  I think that MathematicalProgramResult should offer a string-like sugar getter for that, as well.  (It could conceivably in the future add specific numeric values, instead of just the enum-string.)

After that, maybe we don't need an enum at all?  Or if we do, we can call it something obscure, or use this new SolutionStatus, or etc.

So my proposal is actually:
(1) Remove `MPR::get_solution_result()`.
(2) Keep `SolutionResult` / `MPR::set_solution_result` unchanged -- it's only required for Solve back-end authors to understand, so not a big deal.
(3) Once we finally _remove_ the non-const Solve, then revisit this question.

hongkai-dai, 2019-01-30:
Thanks @jwnimmer-tri . I agree that the enum class (like `SolutionResult`) is too verbose. I will remove `MPR::get_solution_result()` and add `MPR::is_success()` in my next PR.

I think we will need some syntax sugars to retrieve the solution status, when `MPR::is_success() = false`. Typically when the user debugs the optimization program, he/she would want to know why the optimization fails. It could be the solver reaches the iteration limit, the problem is infeasible, a numerical error, etc. Currently that information is stored inside `FooSolverDetails`, like `GurobiSolverDetails::optimization_status`, and the syntax to retrieve this detail is not very straightforward. (Currently we need to do `result.get_solver_details().GetValue<GurobiSolverDetails>().optimization_status`). Namely the user has to know which solver is used, so as to retrieve the details for that solver.

closed 2019-03-25 by hongkai-dai