RobotLocomotion/drake #8403: Ensure testing utilities have negative tests where applicable

EricCousineau-TRI, 2018-03-19:
As a followup from [this comment](https://github.com/robotlocomotion/drake/pull/8396#issuecomment-374305915) in #8396, it'd be useful to have negative tests for testing utilities that do not yet have them.

This link provides some information on how to test these:
https://github.com/google/googletest/blob/master/googletest/docs/AdvancedGuide.md#catching-failures

\cc @sherm1

jwnimmer-tri, 2018-12-21:
I'll propose that we close this under the umbrella of "fix it only when it hurts".  If we find a bug in a test_utilities, we can add a regression test for it then.  I'm not sure we need to go hunt down and test the tests now, nevermind have a reminder issue for it.

EricCousineau-TRI, 2018-12-21:
Agreed, closing.

closed 2018-12-21 by EricCousineau-TRI