RobotLocomotion/drake #8406: Consider decoupling continuous integration from Systems framework

EricCousineau-TRI, 2018-03-20:
Motivated by #8320, it seems that this implementation (single-dimension quadrature realized via an ODE) could be simplified if there were continuous-only implementations of integrators, which integrated via a simple interface (like `f`, `x0`) that were then incorporated into a derived version of `IntegratorBase` via composition rather than inheritance.

\cc @hidmic

EricCousineau-TRI, 2020-04-16:
Closing for now.

closed 2020-04-16 by EricCousineau-TRI

sherm1, 2020-04-16:
This one actually got done -- see [InitialValueProblem](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_initial_value_problem.html)

EricCousineau-TRI, 2020-04-16:
Actually, that's the motivator for this problem that I pointed to in #8320, so I wouldn't say it got done. It still requires wrapping / (over?)encapsulating a more complex interface (`IntegratorBase` with systems) in lieu of having a simpler integrator used by the Systems framework.

That being said, I closed it 'cause it doesn't really seem like a useful thing to pursue atm?