RobotLocomotion/drake #8424: Python DrakeVisualizer

caelan, 2018-03-21:
Hi Drake team,

I'm interested in using Drake's Python bindings. Is there an existing example that populates the drake_visualizer from Python? I found the following test that mocks the visualizer client, but I've been unable to build upon it to display a simulation (even when using DrakeLcm instead of DrakeMockLcm). 
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/multibody/test/rigid_body_plant_test.py

I am able to run the "Basic IIWA Simulation" C++ though.
https://github.com/RobotLocomotion/drake/tree/master/examples/kuka_iiwa_arm

Does using the visualizer from Python require additional bindings for the following classes?
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/lcm_py.cc
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/multibody/rigid_body_plant_py.cc#L169

Thanks,
Caelan

avalenzu, 2018-03-22:
You can find a couple of examples of using `DrakeVisualizer` in Python (these happen to be Jupyter notebooks) at https://github.com/avalenzu/drake-jupyter-examples. Hope that helps!

jwnimmer-tri, 2018-03-31:
@caelan Feel free to ping us back if there are more questions.

closed 2018-03-31 by jwnimmer-tri