RobotLocomotion/drake #8433: 'drake::pydrake::MirrorDef' declared with greater visibility than the type of its field

jamiesnape, 2018-03-22:
```
INFO: From Compiling bindings/pydrake/symbolic_py.cc:
In file included from bindings/pydrake/symbolic_py.cc:13:0:
bazel-out/k8-opt/bin/bindings/pydrake/util/_virtual_includes/wrap_pybind/drake/bindings/pydrake/util/wrap_pybind.h
:17:7: warning: 'drake::pydrake::MirrorDef' declared with greater visibility than the type of its field
'drake::pydrake::MirrorDef::m_' [-Wattributes]
 class MirrorDef {
       ^
bazel-out/k8-opt/bin/bindings/pydrake/util/_virtual_includes/wrap_pybind/drake/bindings/pydrake/util/wrap_pybind.h
:17:7: warning: 'drake::pydrake::MirrorDef' declared with greater visibility than the type of its field
'drake::pydrake::MirrorDef::mirror_' [-Wattributes]
```

EricCousineau-TRI, 2018-03-22:
Gotcha. Seems that I need to inline / hide this class:
http://pybind11.readthedocs.io/en/stable/faq.html#someclass-declared-with-greater-visibility-than-the-type-of-its-field-someclass-member-wattributes

Will see if I can update those flags.

EricCousineau-TRI, 2018-12-18:
Relates #8540 on the level of RTTI and all that.

jwnimmer-tri, 2022-03-14:
Refer to #16756 for a nice trick.  We can set hidden visibility on an entire Drake namespace, instead of fighting with `-fvisibility=hidden` vs the linker.

With the attribute, symbols used within the file but not in the pydrake namespace can still be hidden (or not) per their original declarations.  With the `-fvisibility` build flag all symbols are hidden, which leads to mismatched visibility vs other TUs.

```c++
namespace pydrake __attribute__ ((visibility ("hidden"))) {
...
}  // namespace pydrake
```

jwnimmer-tri, 2023-01-18:
I think this is probably still worth doing, but in terms of scheduling my plan is to do the module flattening first (so that we have fewer python shared libraries -- most likely just one), at which point it should be less of a hassle to get this right.

jwnimmer-tri, 2023-01-27:
See also #18677.  It adds the warning flag, and provides a nice macro for the namespace-level visibility control.

rpoyner-tri, 2026-11-12:
Reassigning to @jwnimmer-tri, the `distribution` lead per the component table in GitHub Issue Management.

Status as I understand it: `class_` and `ParamInit` in `pydrake_pybind.h` are already marked `visibility("hidden")`. Your 2023-01-18 comment planned to flatten the modules first, and pydrake still has per-module `module_py.cc` files.

@jwnimmer-tri, please keep it, relabel it, or close it if you consider it done or not worth doing.