RobotLocomotion/drake #8437: pybind11: Ensure that Bazel `copt` flags are appropriate for the compiler

EricCousineau-TRI, 2018-03-22:
Per @jamiesnape's comment here: https://github.com/robotlocomotion/drake/pull/8435#issuecomment-375417267

> An existing defect, but I saw the following with GCC
> ```
> cc1plus: warning: unrecognized command line option '-Wno-unknown-warning-option'
> cc1plus: warning: unrecognized command line option '-Wno-#warnings'
> ```

I briefly tried using `select(...)` statements, but of course Bazel's really picky about how it parses it and combines it with upstream statements that expect plain lists.

Easiest thing may be render the results of a `select(...)` statement for a compiler into as a `repository_rule`, so that way we can load them as plain variables (similar to @jwnimmer-tri's environment variable tricks).

jamiesnape, 2018-03-22:
> I briefly tried using select(...) statements, but of course Bazel's really picky about how it parses it and combines it with upstream statements that expect plain lists.

You probably hit https://github.com/bazelbuild/bazel/issues/1623.

jwnimmer-tri, 2018-03-22:
We already have `gcc_copts` that you can set: https://github.com/RobotLocomotion/drake/blob/master/tools/skylark/drake_cc.bzl#L312.  I don't think we have `clang_copts` but it could be done in the similar way.

jamiesnape, 2018-03-22:
(which is part of why you were hitting the nested `select`, along with the default compiler-specific flags)

EricCousineau-TRI, 2018-03-22:
Ah, thank y'all! Trying it out `clang_opts` in #8435