RobotLocomotion/drake #8466: kcov errors on mac-sierra

BetsyMcPhail, 2018-03-27:
Lots of tests give the following error: 

> 14:09:05 INFO: From Testing //common/test_utilities:eigen_matrix_compare_test:
> 14:09:05 ==================== Test output for //common/test_utilities:eigen_matrix_compare_test:
> 14:09:05 kcov: warning: kcov: WARNING: kcov has been built without libbfd-dev (or
> 14:09:05 kcov: binutils-dev), so the --verify option will not do anything.
> 14:09:05 
> 14:09:05 Traceback (most recent call last):
> 14:09:05   File "<input>", line 1, in <module>
> 14:09:05   File "/usr/local/Cellar/python@2/2.7.14_3/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy.py", line 52, in <module>
> 14:09:05     import weakref
> 14:09:05   File "/usr/local/Cellar/python@2/2.7.14_3/Frameworks/Python.framework/Versions/2.7/lib/python2.7/weakref.py", line 14, in <module>
> 14:09:05     from _weakref import (
> 14:09:05 ImportError: cannot import name _remove_dead_weakref
> 14:09:05 kcov: error: Launch failure
> 14:09:05 
> 14:09:05 kcov: error: Can't start/attach to common/test_utilities/eigen_matrix_compare_test

BetsyMcPhail, 2018-03-27:
And we also see this error on non-everything builds:

1```
3:23:28 INFO: From Testing //tools/lint:clang_format_includes_test:
13:23:28 ==================== Test output for //tools/lint:clang_format_includes_test:
13:23:28 kcov: warning: kcov: WARNING: kcov has been built without libbfd-dev (or
13:23:28 kcov: binutils-dev), so the --verify option will not do anything.
13:23:28 
13:23:28 kcov: error: Cannot launch process: SBTarget is invalid
13:23:28 
13:23:28 kcov: error: Can't start/attach to tools/lint/clang_format_includes_test
13:23:28 ================================================================================
```

BetsyMcPhail, 2018-03-28:
First error described here: https://stackoverflow.com/questions/47658596/strange-mixing-of-system-homebrew-python-with-lldb

andrewdavidmackenzie, 2019-10-08:
Still happening to me (on mac OS Catalina). Is there a known fix or workaround?

andrewdavidmackenzie, 2020-06-06:
Still trying to understand the root cause of this error on macos:

```
kcov: error: Can't start/attach to /Users/andrew/workspace/flow/target/debug/deps/flowc_execution_tests-112fe8fc3a76dc84.d
kcov: error: Cannot launch process: SBTarget is invalid
```

EricCousineau-TRI, 2020-06-06:
Howdy @andrewdavidmackenzie!
Can you post more info about your env?

Suggestions on what info to provide:
https://drake.mit.edu/getting_help.html#helpful-information

Here are some example steps to provide "reproducbile" information, based off of:
https://drake.mit.edu/bazel.html#kcov

```sh
$ cd drake

# Git info

$ git log -n 1 --oneline
8b0678d408 (HEAD -> master, upstream/master) Add alternative convex Eval functions to LorentzConeConstraint. (#13458)
$ git config remote.upstream.url 
https://github.com/RobotLocomotion/drake.git

# Show that you've re-run install prereqs; please provide any errors that may have occurred
$ ./setup/mac/install_preqreqsh.sh --with-kcov

# Command run that recreated the error? e.g.
$ bazel test --config=kcov //common:polynomial_test
```

If you think environment variables are to blame, please try isolating your environment. On `bash` (Ubuntu), this is possible using `env -i`.

See example here, with `env-isolate`:
https://github.com/RobotLocomotion/drake/issues/13187

jwnimmer-tri, 2020-06-06:
At the moment, Drake does not support kcov on macOS.  Only [these variants](https://drake-jenkins.csail.mit.edu/view/Coverage/) are supported.

jwnimmer-tri, 2020-06-08:
I'll resolve this by updating our documentation to exclude macOS support at https://drake.mit.edu/bazel.html#kcov-on-macos and surrounding text.

jwnimmer-tri, 2020-06-09:
The docs PR is open now.  After that is approved, I'll PR a change to macOS setup to remove kcov as a prerequisite.

For a community discussion for any tips to get macOS kcov working, that seems like a good question to post on StackOverflow (maybe with the "drake" tag).

PR #13522 (2020-06-09): setup: Remove kcov prereq on macOS
Drake does not support kcov build system integration on macOS, per #13521.

Closes #8466.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13522)
closed 2020-06-09 by jwnimmer-tri