RobotLocomotion/drake #8473: GeometrySystem requests

RussTedrake, 2018-03-29:
Not trying to rush... just capturing these for tracking.  I believe this was the primary outcome from the Monday discussion:

- [ ] frame id input port should disappear.  frame id should be combined with pose (and velocity, acceleration) input ports.
- [ ] add support for diffuse color material properties to GeometryInstance (and send to DrakeVisaulizer).
- [ ] Support visualization blocks as a consumer of GeometrySystem
   1) class method that could be called, e.g. from the visualization class constructor, that returns a list of frame id's and geometry instances. 
   2) query method that returns list of frame id's and poses.
- [ ] refactor "GeometrySystem" => "SceneGraph"

SeanCurtis-TRI, 2018-03-29:
@RussTedrake I'm starting on this shopping list. Did you have a proposal for your "visualization" blocks?

The issues as I see it is:
1. What info is actually communicated and how is it organized into structs?
2. As your proposal alludes, there are two operations "declaration" and "update". Probably a unique struct for each.
3. What is the calling mechanism? Both through a QueryObject? Declaration through a `SceneGraph` system method? 

Also, did we want to add the visualization utility system which combines a `SceneGraph` and a the LCM-publishing mechanism compatible with `drake-visualizer` to the shopping list?

sherm1, 2018-03-29:
There was also the issue of needing to register geometry prior to being able to obtain its dimensions from parameters in the Context. I believe @RussTedrake said he had to create the System twice in order to get a usable Context, look at its parameters, and then create a new System using them.

RussTedrake, 2018-03-30:
@sherm1 -- i didn't include that on this short list, because it might be a bigger topic.  I think, longer term, we might need to support runtime registering of geometry.  But i think the current state of affairs is what you would expect given the systems framework.

@SeanCurtis-TRI -- while i'm certainly open to ideas, I chose the words on my wishlist carefully.  i think a class method for getting the initial geometry (e.g. as a list of frame_id, geometryinstance pairs via a struct) and a query via the output port for the poses (again via a std::list?).  Any variations of this are aok for me?

Adding the diagram sugar of SceneGraph+Visualizer sounds great to me, too.

SeanCurtis-TRI, 2018-03-30:
1. Post-allocation geometry registration -- I formerly had an API that allowed exactly this. The difference is it required passing in a context. There's also a subtlety about what it means to allocate a default context. Right now, the default context includes everything registered before initial allocation. So, if I "alloc" context over and over, I'll always get the same scene graph. Would that default allocation change based on subsequent graph changes? Or, would it be better to move *all* registration to post context allocation and the default allocation is *always* empty? We'll have to revisit the design decisions from before.

2. I realize now that I hadn't interpreted your comment the way you'd intended. I didn't read "geometry instance" as `GeometryInstance` and therefore interpreted your original request as a concept and not an implementation. Certainly, it would be possible to do that literally -- a mapping of frame ids to sets of `GeometryInstance`s.  :)

3. I'd still advocate reporting (frame id, pose) pairs as an output port instead of a query. On the one hand, it *is*, strictly speaking, a query and making it part of a query object seems reasonable from that perspective. But, on the other hand:
      - For me, queries are about *operations* on the input poses and, for this particular "query", the operation is just a mux; so simple it's almost *not* an operation.
      - This also hints at the possibility, in the future, of decomposing the single query object into multiple query objects so that we can refine the dependencies at a more granular level. Looking at a diagram, it would be more clear just what kind of dependency lies between a SceneGraph and a consumer system.

RussTedrake, 2018-03-30:
Sounds great to me.

RussTedrake, 2022-02-06:
#16526 will resolve one more of these (SceneGraph support for Expression).

jwnimmer-tri, 2023-07-30:
It's not clear to me what work (if any) remains here.

We already have API to interrogate the scene graph (query object, geometry inspector), to change shapes, and to add geometries.

What's still missing?  This issue doesn't seem actionable, currently.

Perhaps opening a more specific issue for each remaining missing item (and closing this issue) would help clarify.

RussTedrake, 2023-07-30:
I agree.  The ideas here are stale at best.

closed 2023-07-30 by RussTedrake