RobotLocomotion/drake #8475: Bazel is using user site packages by default

jwnimmer-tri, 2018-03-29:
Something like
- https://docs.python.org/2/library/site.html#site.USER_SITE
or
- https://docs.python.org/2/library/site.html#site.USER_BASE

is enabled by default when Bazel is running python stuff, which means that `$HOME/.local` can non-hermetically enter the build.

This is bad in general, but specifically bad for genrules such as https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/godotengine/main_make_binders.py where we do `import core` and some other thing in `.local` was named `core`, and then _poof_ our genrule fails.

Note that this is _different_ than `PYTHONPATH` leaks -- I think we are correctly guarding against PYTHONPATH breaking the hermetic seal.  The user site mechanism appears to operate distinctly and outside of the PYTHONPATH mechanisms.

More details / fixes to follow as I explore more.  This is just a placeholder for now.

jwnimmer-tri, 2020-03-16:
If I'm recalling correctly, #8476 fixed tools (genrules, actions, etc.) that use python, but did not isomtae things like unit tests, which still use user site-packages.

Now that we have platform-specific bazel rcfiles, I believe we _could_ disable `$HOME` on Ubuntu only.  Seems like we _should_?  WDYT?

EricCousineau-TRI, 2020-03-16:
Hm... What if we did something like adding `test --test_env=PYTHONNOUSERSITE=1` to `ubuntu-bionic.bazelrc`?

jwnimmer-tri, 2020-03-16:
Yes, that's essentially what I was proposing.  Though we'd need to check if `--test_env` causes too much flux when alternating `bazel build //...` vs `bazel test //...`.  (I see to recall it re-runs the analysis phase if you use `test --test_env`, should use `build --test_env` instead possibly.)  We may need to use `--action_env` as well / alternatively.

jwnimmer-tri, 2020-03-18:
FYI I'm prototyping the rcfile changes in Anzu.  If we bake that for a while and like it, I'll pull the changes into Drake (Ubuntu only) next.

jwnimmer-tri, 2020-04-29:
We've had this in TRI (Anzu) since March.  It seems to be working great:
```ini
# Disable python imports from ~/.local (pip -U) during build and test.
# https://github.com/bazelbuild/bazel/issues/4939
# https://github.com/RobotLocomotion/drake/issues/8475
build --action_env=PYTHONNOUSERSITE=1
build --test_env=PYTHONNOUSERSITE=1
```

We should probably add it to our Ubuntu rcfile in Drake.

PR #8476 (2018-03-31): Mitigate python user site-packages
Closes #8475 as best we can right now.  Hopefully upstream will provide a more comprehensive solution.

Do not merge until:
- [ ] macOS builds pass.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8476)
PR #13190 (2020-05-01): tools: Disable python imports from ~/.local (pip -U)
On Ubuntu, we purposefully do not use pip.  By removing the homedir from imports, we can ensure more reproducible builds and tests.

Closes #8475.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13190)
closed 2020-05-01 by jwnimmer-tri